arduino/arduino-cli #1264: esp32:esp32:esp32 linking fails

FStefanni, 2021-04-13:
## Bug Report

### Current behavior

Hi,

the following sketck, which is the standard blink example slightly modified, fails to link.

```c++
#define LED_BUILTIN 11
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(115200);
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  String s = "Hohoho";
  Serial.println(s);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

These are the results:
* Compiling for ESP32 Dev board via Arduino IDE: success
* Compiling for ESP32 Dev board via arduino-cli (`arduino-cli compile -v --fqbn esp32:esp32:esp32 Blink.ino`): linking fails
* Compiling for ESP32 WROVER via arduino-cli, (`arduino-cli compile -v --fqbn esp32:esp32:esp32wrover Blink.ino`): success

So it seems only the esp32:esp32:esp32 board fails with arduino-cli (but it is fine if compiling via IDE).

The output of the linking is something like this:

```
/home/kappa/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o):(.literal.main_task+0x14): undefined reference to `app_main'
/home/kappa/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o): In function `main_task':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:545: undefined reference to `app_main'
/tmp/arduino-sketch-48A3503F6DEB4CDDCA880B60FB16D10F/sketch/Blink.ino.cpp.o:(.literal.startup._GLOBAL__sub_I__Z5setupv+0x4): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
/tmp/arduino-sketch-48A3503F6DEB4CDDCA880B60FB16D10F/sketch/Blink.ino.cpp.o: In function `_GLOBAL__sub_I__Z5setupv':
/home/kappa/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/IPAddress.h:94: undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
```

### Expected behavior

It should link successfully :)

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.18.0 Commit: 46c23166 Date: 2021-03-31T09:42:54Z
- OS and platform: the same results happen on both Linux (Ubuntu 20.04) and Windows 10.

### Additional context

I am hot sure if this is a issue of arduino-cli or the ESP32 package, but since it works fine with the IDE, I suppose it is a CLI bug.

Regards

FStefanni, 2021-04-30:
With the newer version it now works:

arduino-cli alpha Version: 0.18.1 Commit: b3cf8e19 Date: 2021-04-13T13:08:30Z

closed 2021-05-11 by silvanocerza

rhl-, 2021-08-24:
Hi, I am experiencing the same problem, with 0.18.1 or 0.18.3. 

```
pi@raspberrypi:~ $ arduino-cli compile --fqbn esp32:esp32:esp32 GetChipId
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o):(.literal.main_task+0x14): undefined reference to `app_main'
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o): In function `main_task':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:545: undefined reference to `app_main'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0x4): undefined reference to `String::String(char const*)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0x8): undefined reference to `String::concat(char)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0xc): undefined reference to `String::~String()'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `Stream::readString()':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::String(char const*)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `String::operator+=(char)':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::concat(char)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `Stream::readString()':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::~String()'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
pi@raspberrypi:~ $ arduino-cli version
arduino-cli alpha Version: 0.18.1 Commit: b3cf8e19 Date: 2021-04-13T13:08:30Z

pi@raspberrypi:~ $ curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | sh 
Installing in /home/pi/bin
ARCH=ARMv7
OS=Linux
Using curl as download tool
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.18.3_Linux_ARMv7.tar.gz
arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-14T12:36:58Z installed successfully in /home/pi/bin
pi@raspberrypi:~ $ arduino-cli compile --fqbn esp32:esp32:esp32 GetChipId
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o):(.literal.main_task+0x14): undefined reference to `app_main'
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/lib/libesp32.a(cpu_start.o): In function `main_task':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:545: undefined reference to `app_main'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0x4): undefined reference to `String::String(char const*)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0x8): undefined reference to `String::concat(char)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o):(.literal._ZN6Stream10readStringEv+0xc): undefined reference to `String::~String()'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `Stream::readString()':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::String(char const*)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `String::operator+=(char)':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::concat(char)'
/tmp/arduino-core-cache/core_esp32_esp32_esp32_e39513a897a43c8d73dca388c2d479a7.a(Stream.cpp.o): In function `Stream::readString()':
/home/pi/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Stream.cpp:336: undefined reference to `String::~String()'
collect2: error: ld returned 1 exit status

Error during build: exit status 1
```

FStefanni, 2021-08-24:
Hi,

this is a mysterious bug, since it only happens sporadically, and only by using arduino-cli.
For me, everything works fine since when I closed this bug.

I am wondering if maybe it is something depending on the creation of the static libs (.a), 
which maybe are not "updated/refreshed" correctly...

Try to just delete completely the build directory (/tmp/arduino-core-cache) to check if it fixes.

Regards.

rhl-, 2021-08-24:
I can try this. Would you mind providing some commentary on how to get more verbosity on what the tool is doing? Which compile commands is it executing? Etc? Perhaps this is in the cache?

FStefanni, 2021-08-24:
Hi,

I am not an expert on this tool usage. i just know `-v` option (the verbose option).

Regards.

silvanocerza, 2021-08-24:
I can't reproduce this on Linux with all `0.18.x` versions, it compiles sucessfully for both boards using `esp32:esp32@1.0.6`. 

@FStefanni could you provide the full output in any case? It might be helpful.

@cmaglie what do you think of this?

FStefanni, 2021-08-25:
Hi,

sorry, I cannot, since I already closed this issue a lot of time ago: for me it now works, and I never faced this issue again
(nor I am able to reproduce it now).

At that time, I thought that maybe it was just a strange/unexpected setting in my PC, since this is a very strange issue,  and
since switching to a newer version of `arduino-cli` fixed it, and no one had the same issue or was able to reproduce it.

But now also @rhl is having the same issue I had: very strange.

All I can think is that it could depend on the creation of the static libs... maybe some wrong timestamp ("clock skrew") is not correct, and prevents to re-compile/update them correctly. Or maybe some memory issue ("invalid read/write") of `arduino-cli`
(in `C++` this is the most common cause of mysterious bugs, but I do not know if this is possible in `go`).

Regards

silvanocerza, 2021-08-25:
Ok, thanks.

I'll reopen the issue if there are further development on it.

RodrigoDornelles, 2022-05-27:
had the problem in 0.22