arduino/arduino-cli #1269: The type declared after any function is not visible as the parameter type for any other function below.

allnick, 2021-04-18:
Hi.
There is a problem.
The type declared after any function is not visible as the parameter type for any other function below.
Although it is suitable for other purposes.
For example:

```cpp
struct tword{
  byte low;
  byte high;
};

void none(){}

bool test(tword par){return true;}

void setup() {
  word w;
  tword* var = (tword*)(&w);
}

void loop(){}
```

It's ok.

```cpp
void none(){}

struct tword{
  byte low;
  byte high;
};

//bool test(tword par){return true;}

void setup() {
  word w;
  tword* var = (tword*)(&w);
}

void loop(){}
```

Too it's ok

```cpp
void none(){}

struct tword{
  byte low;
  byte high;
};

bool test(tword par){return true;}

void setup() {
  word w;
  tword* var = (tword*)(&w);
}

void loop(){}
```

What a hell !!!?
'tword' was not declared in this scope !!!

per1234, 2021-04-18:
Hi @allnick. Thanks for your report.

During [sketch preprocessing](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#pre-processing), the Arduino build system (which is implemented here in the Arduino CLI code base) automatically generates function prototypes in .ino files for any functions that don't already have a prototype. This turns out to be surprisingly difficult. Although the prototype generation system works fine for most sketches, under certain circumstances, it inserts the prototype at the wrong position in the code. That is the cause of the error you encountered.

You can see how the code looks after preprocessing here:
```
$ arduino-cli compile --fqbn arduino:avr:uno --preprocess Foo
#include <Arduino.h>
#line 1 "C:\\Foo\\Foo.ino"
#line 1 "C:\\Foo\\Foo.ino"
void none();
#line 8 "C:\\Foo\\Foo.ino"
bool test(tword par);
#line 10 "C:\\Foo\\Foo.ino"
void setup();
#line 15 "C:\\Foo\\Foo.ino"
void loop();
#line 1 "C:\\Foo\\Foo.ino"
void none(){}

struct tword{
  byte low;
  byte high;
};

bool test(tword par){return true;}

void setup() {
  word w;
  tword* var = (tword*)(&w);
}

void loop(){}
```
Note that the `bool test(tword par);` prototype was inserted above the declaration of the `tword` type.

The workaround is to simply manually add your own prototype in the correct location:
```cpp
void none(){}

struct tword{
  byte low;
  byte high;
};
bool test(tword par); // function prototype
bool test(tword par){return true;}

void setup() {
  word w;
  tword* var = (tword*)(&w);
}

void loop(){}
```

allnick, 2021-04-18:
Respected per1234. Thanks for your explanation.
But I just moved the type declaration above all the functions.
But your remark, I necessarily take into account.

per1234, 2024-09-02:
Thanks for your report @allnick. I see we have another report of the bug at https://github.com/arduino/arduino-cli/issues/2696. Since the other one was submitted earlier, and it a bit more detailed, I'll close this in favor of the other.

closed 2024-09-02 by per1234