arduino/arduino-cli #1271: Allow specifying arbitrary config files on the commandline

matthijskooijman, 2021-04-19:
## Bug Report

### Current behavior

Currently, there are only a few items from arduino-cli.yaml that can be specified on the commandline:

https://github.com/arduino/arduino-cli/blob/e5f4c95a6a081cee1fea876cdcb6d2c330a6b2da/configuration/configuration.go#L74-L79

To specify other options, you must create an arduino-cli.yaml file (in the current dir or pass it to `--config-file`). When running from a build script, creating such a file (and cleaning it up again) is sometimes cumbersome, it would be easier to be able to specify arbitrary values from the commandline instead.

### Expected behavior

One way would be to add commandline optinos for each value in the config. This would provide nice help output, but also overload the help output and be prone to forgetting to add new config file options to the commandline.

An easier alternative would be to just allow specifying configuration items by name directly on the commandline. E.g.:

```
arduino-cli --config directories.user=/path/to/sketchbook compile
```

Note that for my particular usecase (an automatic build script that needs to set the sketchbook/user dir to make specific hardware and libraries available) I just need to set `directories.user`, but it is probably nicer to make this generic.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: git-snapshot Commit: 021b4496 Date: 2021-04-19T13:35:49Z
- OS and platform: Ubuntu Focal, x86_64.

silvanocerza, 2021-05-10:
I feel having a flag for each config would be a bit too cumbersome to tell you the truth.

If you need to set different configs for different executions of the ArduinoCLI you can use env vars, all settings can be set prefixing them with `ARDUINO_`, so to set `directories.user` you can just do this:

```
ARDUINO_DIRECTORIES_USER=/path/to/sketchbook arduino-cli compile
```

umbynos, 2022-12-14:
I agree with Silvano,
If you feel that there's some nice use case, we could add a flag maybe

closed 2022-12-14 by umbynos