arduino/arduino-cli #1273: Could not add proxy setting to the configuration file thanks the cli

fpistm, 2021-04-21:
## Bug Report

arduino-cli can managed the proxy setting in its configuration file: `arduino-cli.yaml`
Like this:
```
network:
  proxy: http://<login>:<password>@myproxy.ext:port
```

Anyway it seems it does not able to add it thanks the cli only by editing manually the file.

### Current behavior

`arduino-cli config add network.proxy http://<login>:<password>@myproxy.ext:port`

Result to `Settings key doesn't exist`

### Expected behavior

Be able to add the proxy configuration thanks the cli.

### Environment
- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.18.1 Commit: b3cf8e19 Date: 2021-04-13T13:08:30Z
- OS and platform: Linux

### Additional context
In a general way all possible setting keys should be supported thanks `arduino-cli config <set|add> key.name key.value` commands.

Thanks

silvanocerza, 2021-05-10:
Yep, I forgot to add the `network.proxy` config to the list of valid ones, thanks for the report.

closed 2021-05-13 by silvanocerza

fpistm, 2021-05-13:
Thank you @silvanocerza

silvanocerza, 2021-05-13:
:pray: 

This will be available in the next release for sure, no idea when though.

PR #1288 (2021-05-13): Fix `network` settings uneditable with config command
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking chan [... cut]