arduino/arduino-cli #1279: macOS Catalina - Error updating index: invalid handle

bnwlkr, 2021-05-06:
I installed arduino-cli via homebrew on macOS Catalina. When I tried running `arduino-cli core update-index` as listed in the Getting Started, I got `Error updating index: invalid handle`. I couldn't find anything helpful with Google so I downloaded the source and added print statements. I found that the issue was just that I didn't have Full Disk Access enabled for Terminal (see [here](https://osxdaily.com/2018/10/09/fix-operation-not-permitted-terminal-error-macos/)). Just leaving this here in case someone else comes across the same issue. For the devs, the 'operation not permitted' error message should probably be included in the output here.

ubidefeo, 2021-05-07:
@silvanocerza can we look into this?

silvanocerza, 2021-05-07:
Thanks @bnwlkr, I'll reopen this so we can keep track of it.

This is probably just a matter of showing a better error to the user, it should be said clearly that more permissions are needed. In this case it probably failed because it couldn't create the necessary `Arduino15` folder.

MatteoPologruto, 2023-01-30:
As of now ([0.29.0](https://github.com/arduino/arduino-cli/releases/tag/0.29.0)), a specific error is shown when the CLI fails to create the `Arduino15` directory.
```
// Setup data directory
dataDir := configuration.DataDir(configuration.Settings)
packagesDir := configuration.PackagesDir(configuration.Settings)
if packagesDir.NotExist() {
	err := packagesDir.MkdirAll()
	if err != nil {
		return nil, &arduino.PermissionDeniedError{Message: tr("Failed to create data directory"), Cause: err}
	}
}
```
This is handled inside the [Create](https://github.com/arduino/arduino-cli/blob/master/commands/instances.go#L148) function, which is part of the initialization process that starts when `core update-index` is executed for the first time.
For this reason, I'm closing the issue.

closed 2023-01-30 by MatteoPologruto