arduino/arduino-cli #1289: Arduino IDE reads CPLUS_INCLUDE_PATH when it shouldn't

apexys, 2021-05-14:
Hi,
I've just done a fresh install of the Arduino IDE version 1.8.14 on my system and when I tried to compile a blink sketch, it gave me the following error:

```
In file included from C:\msys64\mingw64\include\c++\10.2.0/stdlib.h:36:0,
                 from C:\Users\Valentin Buck\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:23,
                 from sketch\Blink.ino.cpp:1:
C:\msys64\mingw64\include\c++\10.2.0/cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Arduino Uno.
```

It looks like avr-gcc tries to include a file from my mingw64-installation instead of the arduino specific headers.

If I remove the environment variable `CPLUS_INCLUDE_PATH`, the blink sketch compiles normally.

Is there a better way to do this?

Thanks!

matthijskooijman, 2021-05-14:
Hm, it seems like gcc is being influenced by environment variables, as documented here: https://gcc.gnu.org/onlinedocs/cpp/Environment-Variables.html

I guess a potential bug here is that arduino-cli/arduino-builder runs gcc with original environment variables, while it actually aims to strictly control all gcc settings, so it should probably unset these env vars. On the other hand, one could also argue that, by setting the `CPLUS_INCLUDE_PATH` env var, you as a user are instructing the IDE and its gcc to do something different, and if that's not what you want, then you shouldn't set that env var.

Assuming that we think this is a bug in Arduino and it should be fixed there, I can imagine three options:
 1. Always run all compile (and upload) processes in a clean / empty environment. This might be tricky, because an empty env (i.e. no PATH) probably does not work and I'm not sure if there is a portable way to generate a clean/minimal env. Also, I suspect that there are env vars that *are* expected to be passed on, maybe some other settings for gcc, upload tools or platform scripts are set in env vars that should be respected, then running with a clean env breaks those.
 2. Unset just the list of gcc env vars from the above documentation page. Caveats are that the list might change in the future, and it's actually not up to arduino-cli to hardcode this list, since it is not even sure that the compiler is gcc (this is all platform-defined). This is probably the most pragmatic option, though.
 3. Allowing platform.txt to specify a list of env vars to clear (or maybe set). This is the cleanest approach, but does mean that this must be fixed in all platforms separately, meaning that in practice a lot of the third-party ones will probably remain broken.

I guess option 2. is the best way forward, or declaring that this is a user problem, not a bug.

What you might do to fix this right now, is to use CPATH instead of CPLUS_INCLUDE_PATH, since that adds include paths at the very end of the list instead of before the default system include paths. It might not work for your mingw setup, though, since it adds them even after any commandline `-I` options, which might be too late.

Note that I moved this issue to the arduino-cli repo, since that's where the code lives that runs the build process, and that's where these changes need to be.

per1234, 2021-05-14:
The `GCC_EXEC_PREFIX` environment variable has also been the cause of some similar suffering for Arduino users:
https://gcc.gnu.org/onlinedocs/gcc/Environment-Variables.html#index-GCC_005fEXEC_005fPREFIX