arduino/arduino-cli #1297: AVRISP mkII not detected

jeffrson, 2021-05-21:
## Bug Report

### Current behavior

```
C:\arduino-1.8.13\arduino-cli upload -P avrispmkii -i .\sketch.ino.hex -b custom:board
```

`Error during Upload: programmer 'avrispmkii' not available`

### Expected behavior

Should flash mentioned hex file, just like avrdude does:

```
C:\arduino-1.8.13\hardware\tools\avr/bin/avrdude -CC:\arduino-1.8.13\hardware\tools\avr/etc/avrdude.conf ^
-v -pcustom -cstk500v2 -Pusb -Uflash:w:sketch.ino.hex:i
```

This command has been copied from Arduino IDE logs (just blanked out the actual board and sketch path).

### Environment

- CLI version (output of `arduino-cli version`): 0.18.2
- OS and platform: Windows 10 20H2

### Additional context

I made AVRISP mkII work with USB drivers from
https://www.microchip.com/mplab/avr-support/avr-and-sam-downloads-archive modified to libusb-win32 with https://zadig.akeo.ie.

matthijskooijman, 2021-05-21:
@jeffrson, which custom board are you using?

Recently, the behavior of programmer selection in the Java IDE was changed to only offer programmers defined by the core that the selected board is in (or a core referenced by it). This was introduced to prevent showing e.g. AVR-based programmers when working with a SAMD board, or even within the same architecture, showing programmers from another platform that would not work because they need different information.

If arduino-cli implements this the same way (haven't checked), and your custom core does not offer any programmers, nor references the arduino:avr core, then that might explain why the programmer is not available.

jeffrson, 2021-05-21:
I made a new "platform" for xmega256a3u based on https://github.com/Xmegaduino/Xmegaduino.

Yes, Arduino IDE doesn't show the programmer, but avrdude just works.

> Recently, the behavior of programmer selection in the Java IDE was changed to only offer programmers defined by the core that the selected board is in (or a core referenced by it). 

Do you have a reference how I could add my board to the programmer?

Thanks for pointing this out!

per1234, 2021-05-21:
Hi @jeffrson the programmers are defined in a configuration file named `programmers.txt`, which is placed in the same folder as your platform's boards.txt file. You can learn about it here:
https://arduino.github.io/arduino-cli/latest/platform-specification/#programmerstxt

And you can see the official programmers.txt definition for the AVRISP mkII here:
https://github.com/arduino/ArduinoCore-avr/blob/master/programmers.txt#L8

Since this is the expected behavior when attempting to use a programmer that is not defined by the board's platform, I'll go ahead and close this issue.

closed 2021-05-21 by per1234

matthijskooijman, 2021-05-21:
> Do you have a reference how I could add my board to the programmer?

Sounds like your core code support all AVR-based programmers, so the easiest path forward is probably to just copy the AVR programmers.txt that @per1234 linked above into your core verbatim. There is currently no way to import programmers from another platform automatically, except when you use the actual core (the C/C++ code) from another platform verbatim using a referfence.

jeffrson, 2021-05-21:
Yes, indeed - that's just what I did (copy programmers.txt to my custom platform).

Now the Arduino IDE again shows the list of programmers - and VSCode Arduino extension does this as well (failed obviously for the same reason). That's great!

Finally, I can confirm that
`C:\arduino-1.8.13\arduino-cli upload -P avrispmkii -i .\sketch.ino.hex -b custom:board`
now works for me.

Thanks a lot!