arduino/arduino-cli #1306: Using --git-url fails to install library due to repository and header name mismatch

hawk4031, 2021-06-03:
## Bug Report

### Current behavior

`arduino-cli lib install --git-url https://github.com/ThingPulse/esp8266-oled-ssd1306`

When running the above command, I receive an error: `Error installing Git Library: library is not valid: missing header file "esp8266-oled-ssd1306.h"`

This particular library doesn't have a header file named that. What is interesting though is if I instead install the library by the name, then it installs exactly the same repository with no issues. Here is that command:

`arduino-cli lib install "ESP8266 and ESP32 Oled Driver for SSD1306 displays"`

I verified it was the exact same library by using `arduino-cli lib search`

### Expected behavior

When pulling using Git repositories, it should install the libraries the same way it does when specifying by name.

### Environment

- CLI version (output of `arduino-cli version`): 0.18.3
- OS and platform: Windows for testing, Linux for Travis CI

### Additional context

In previous versions of the CLI, we could specify the Git repository URLs with no problems. We specify quite a few libraries via the CLI in a Travis build. About half of them would need to be replaced with the name as found via `arduino-cli lib search` and there are a few that don't even show up in the library search, so the only way to retrieve them would be via the Git repository, which doesn't work because the name doesn't match.

chmanie, 2021-08-23:
The same happened to me when trying to install [`Control-Surface`](https://github.com/tttapa/Control-Surface) this way. The header file there is called `Control_Surface.h` instead of the expected `Control-Surface.h`. Would there be a way to make that configurable?

per1234, 2021-08-23:
It's not a matter of making it configurable. It's a matter of fixing the bug in Arduino CLI.

There is no requirement for the header file name to match the library repository name/folder name, so the Arduino CLI should  install this library with no problem.

chmanie, 2021-08-24:
Can I help somehow? I'm happy to provide a PR, I just don't know from what angle to approach this. Should the library just be copied to the libraries folder or are there more steps involved? Can this check just be removed and everything will work?

silvanocerza, 2021-08-24:
@chmanie help is always more than welcome and encouraged. 

We probably can just check if the `library.properties` file exists in the root of the Git repository to verify that it's actually a library and not just a generic repository. 

For reference this check has been introduced with #1234, probably I was way to strict to expect an header with the same name of the library.

@per1234 @ubidefeo @cmaglie can any of you confirm that's enough to check for a `library.properties` file?

per1234, 2021-08-24:
It should support installation of [1.0/"legacy" format libraries](https://arduino.github.io/arduino-cli/latest/library-specification/#old-library-format-pre-15) that don't have a `library.properties` file.

For those libraries, a header file must be present in the library's root folder.

chmanie, 2021-08-24:
@per1234 Does the header file have to follow any naming conventions? Or should it just be the first (or only?) header file that is found?

silvanocerza, 2021-08-24:
It must contain at least one header file in the root folder, the naming is not important. 

To make the verification step work correctly you must first verify if the `library.properties` exists, if it does you're good, if it doesn't search for an header file in the root, if there's at least one you're good else that's not a library.

FStefanni, 2021-09-20:
Hi,

I am also interested in this issue.
Probably it is not required to have a header file in the root directory, since Arduino seems to allow a `src` subdir.
But I am not sure about the actual specifications of the different versions.

This is for sure a very critical bug/issue, since it blocks the installation of many libraries.
Or, conversely, the Arduino specification could introduce a braking change... but I do not believe this is the best direction.

Regards.

closed 2021-09-24 by silvanocerza