arduino/arduino-cli #1313: Add native support for Apple Silicon

Logicer16, 2021-06-10:
### Current behaviour
Arduino CLI runs under Rosetta 2 on Apple Silicon. While this works at the moment, it is more efficient (and will eventually be required) to run a native version.

### Expected behaviour
Arduino CLI has a native Apple Silicon version.

### Environment
- CLI version (output of `arduino-cli version`): ```arduino-cli alpha Version: nightly-20210610 Commit: 16f41352 Date: 2021-06-10T01:26:29Z```
- OS and platform: macOS Big Sur 11.5 Public Beta 2, M1 Mac mini

Kompost14, 2021-08-29:
Cause of the missing native support, the "normal"  'make setup' of [Kaleidoscope](https://github.com/keyboardio/Kaleidoscope) fails:

```
$ make setup
...
...
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.18.3_macOS_arm64.tar.gz
Did not find a release for your system: macOS arm64
Trying to find a release using the GitHub API.
LATEST_RELEASE_URL=https://api.github.com/repos/arduino/arduino-cli/releases/tags/0.18.3
Getting https://api.github.com/repos/arduino/arduino-cli/releases/tags/0.18.3
Sorry, we dont have a dist for your system: macOS arm64
You can request one here: https://github.com/arduino/arduino-cli/issues
Failed to install arduino-cli
make: *** [/Users/***/Development/GitHub/Kaleidoscope/bin/arduino-cli] Error 1
```

Thanks for the hint with Homebrew.

brentru, 2022-02-14:
Hi, any update on Apple Silicon Support?