arduino/arduino-cli #1314: Missing STM32F4 pin definitions

bobemoe, 2021-06-10:
## Bug Report

Using  https://github.com/stm32duino/Arduino_Core_STM32 it seems the pin definitions are not loaded properly.

### Current behavior

```
./arduino-cli compile -b STMicroelectronics:stm32:GenF4`

Test.ino:20:17: error: 'PC15' was not declared in this scope; did you mean 'PC_15'?
   20 | #define TFT_CS  PC15
      |                 ^~~~
```

### Expected behavior

The sketch should compile as it does (successfully) in the ArduinoIDE.  PC15 is a valid pin number in the IDE.

### Environment

* arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-14T12:36:58Z
* Linux Devuan 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux

bobemoe, 2021-06-10:
I notice in the IDE there is option "Board Part Number" where I select BlackPillF411CE.  I have not found similar option in the CLI.  I wonder if this contains the definitions?

matthijskooijman, 2021-06-10:
> I notice in the IDE there is option "Board Part Number" where I select BlackPillF411CE. I have not found similar option in the CLI. I wonder if this needs to be set to specify the correct definitions?

This is likely the problem indeed. You can add additional options to the `-b` FQBN:

```
-b  STMicroelectronics:stm32:GenF4:pnum=BLACKPILL_F411CE
```

IIRC you can add additional of these "board options" by separating them with a comma `,` (or maybe another colon `:`).

The exact options available depend on the core and board selected (and the STM32 core recently changed all these options I think, so doublecheck which applies to your version). To see available options, use e.g.:

```
arduino-cli board details STMicroelectronics:stm32:GenF4
```

If this solves your problem, then I think this can be closed. If it does not, then it is likely a problem with the STM32 core and should probably be reported at their repository, so could be closed here as well. So I'm going to go ahead and close this issue already, but let us know if you fixed your issue, or when you report an issue at the STM32 repo?

closed 2021-06-10 by matthijskooijman

bobemoe, 2021-06-10:
Thanks for your speedy reply :D

Yes, this fixed the issue.

Sorry I missed the `board details` option. Its always the one place you don't look :/