arduino/arduino-cli #1315: Feature request - Allow additional location for libraries, for libraries under development.

MHotchin, 2021-06-12:
Right now, libraries under development need to reside in the 'libraries' folder under the 'sketchbook' location.

This works well enough for initial development, but is problematic for continuing development.  For example, I have a local fix in the 'Ethernet' library.  If a new official version comes out, then merging the new version with the work in progress is difficult, since the IDE may stomp on the working copy I have.  Instead, the merge should come through git and propagated to my machine that way.  Essentially, having two entities trying to update the same location using entirely different means is a recipe for disaster.

If in-development libs could reside elsewhere, continuing development would be much easier. I envision:
 - A new folder that is seached during the build process.  Perhaps a message during build that in-development libs are being used.
 - the IDE never updates libs in this location.
 - It is of lower priority than the one in the sketchbook, ie the library manager version would need to be removed before the in-development version was used.
 - As an 'advanced' feature, no need to make IDE UI changes - document it in the library development web pages.

silvanocerza, 2021-06-14:
With the `arduino-cli` you can specify additional libraries folders or specify a path to a library when running the `compile` command.

The `--libraries` flag expects a path to a folder containing multiple libraries, much like the `<sketchbook>/libraries` folder does.
Assume we have two libraries in these paths:
```
/home/user/workspace/libraries/WiFi101
/home/user/workspace/libraries/AudioZero
```
The `--libraries` flag must be used like so:
```
arduino-cli compile -b arduino:avr:uno --libraries /home/user/workspace/libraries /path/to/MySketch
```

Instead the `--library` flag expects a path to the root folder of a single library, so it must be used like so:
```
arduino-cli compile -b arduino:avr:uno --library /home/user/workspace/libraries/WiFi101 --library /home/user/workspace/libraries/AudioZero /path/to/MySketch
```

The library set with `--library` gets top priority if there is another with an identical name.

There is more information in PR #1258.

per1234, 2022-03-31:
There is a similar request at https://github.com/arduino/arduino-cli/issues/1257

However, this request is different in that the other is about supporting organization of libraries into arbitrary folder structures, which would be searched recursively during library discovery but not be an intentional factor in [resolution priority](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#dependency-resolution). In this case, it would be a folder of a specific name which is given special treatment in terms of `lib install` and dependency resolution.

umbynos, 2022-10-20:
Hi @MHotchin can we consider [this](https://github.com/arduino/arduino-cli/issues/1315#issuecomment-860538455) solved using this trick?

MHotchin, 2022-10-21:
> Hi @MHotchin can we consider [this](https://github.com/arduino/arduino-cli/issues/1315#issuecomment-860538455) solved using this trick?

It addresses the problem for command line users, but doesn't appear to help IDE users.

I'll leave it to you to decide if the IDE needs to be supported, and if you decide not, then close the issue.

umbynos, 2022-10-21:
Thanks! I'll close this one then since this is not strictly related to the CLI. Feel free to open a feature request in the [Arduino IDE repository ](https://github.com/arduino/arduino-ide)

closed 2022-10-21 by umbynos