arduino/arduino-cli #1317: Template file extensions are not supported in sketches

per1234, 2021-06-14:
Files with .tpp or .ipp extension in the sketch folder are not copied to the temporary build folder.

Reference: https://github.com/oxullo/Arduino-MAX30100/issues/46

silvanocerza, 2021-06-17:
@ubidefeo @cmaglie 
Do we want to add support for this extension? There is a recent PR adding it #1316 but am not sure if we really want to do this.
What about the Java IDE? I guess we'd have to do it there too? Or maybe not? :eyes:

tshcherban, 2021-06-21:
@silvanocerza is there some particular problem why not add those files support?
I'm often putting libraries into 'src' subfolder within a sketch directory, and some of them are written with modern c++ features including .tpp
IDE support is not necessary, since library file is not edited so often, but compilation would be nice.

silvanocerza, 2021-06-21:
No particular problem really, just wanted to ask feedback to others since it's a long standing issue and there might be reasons it has not been done already. 

As far as I know C and C++ don't have specifications for file extensions, you could even omit them I think. Nonetheless I understand it might be useful to add support for other extensions. 

My main concern is that adding this one would mean having to release a new version of the Java IDE to support it there too, I understand you might not need it but I would very much prefer if we keep them in sync for now.

tshcherban, 2021-06-21:
OK, i see. Just was not very familiar with IDE repo, thought that this list (of extensions) is shared and reused in IDE Gui code.

silvanocerza, 2021-06-21:
Nope, only the new IDE 2.0 uses the same extensions since it uses the CLI as its backend, the Java IDE supported extensions are defined here:

https://github.com/arduino/Arduino/blob/a55a6d2a9d62b7bddbd9dbe9de784cbbbf6540a3/arduino-core/src/processing/app/Sketch.java#L19-L23

silvanocerza, 2021-06-22:
Fixed by #1316.

closed 2021-06-22 by silvanocerza