arduino/arduino-cli #1318: Please, remove the Monitor-Memorial-Minute

ullix, 2021-06-14:
Using Arduino IDE 2.0 Beta7 on ESP32-Dev module:

When you have checked both Compile and Upload under File->Preferences and click Upload you see this output when the compilation finishes:
```
Compilation complete.
Waiting for upload port...
```
Then follows a 10.0 sec pause with nothing happening. I call this the Monitor-Memorial-Minute. Then it continues with:
```
No upload port found, using /dev/ttyUSB2 as fallback
python "/home/ullix/.arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool.py" --chip esp32 --port "/dev/ttyUSB2" --baud 921600  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 "/home/ullix/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/partitions/boot_app0.bin" 0x1000 "/home/ullix/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/bin/bootloader_qio_80m.bin" 0x10000 "/tmp/arduino-sketch-28E92FB090FA4418BDCD7C71BD73CA95/main.ino.bin" 0x8000 "/tmp/arduino-sketch-28E92FB090FA4418BDCD7C71BD73CA95/main.ino.partitions.bin"
esptool.py v3.0-dev
Serial port /dev/ttyUSB2
Connecting....
Chip is ESP32-D0WD-V3 (revision 3)    
...
```
At the same time the menu shows this:
[image]

So, the IDE is well aware that the port to use is properly defined (here: /dev/ttyUSB2), yet, it tells to be waiting for an "upload port", which is not defined and cannot be defined anywhere, and after 10 sec sharp it successfully continues with the properly defined port, which it was supposed to use anyway, but now calls it a "fallback".  What is the point?

As long as there is no extra upload port, different from the default port, please remove this unnecessary waiting.

It is annoying when compilation and upload time almost double for no purpose! Thanks.

per1234, 2021-06-14:
>So, the IDE is well aware that the port to use is properly defined (here: /dev/ttyUSB2), yet, it tells to be waiting for an "upload port"

The reason is that the "ESP32 Dev Module" board definition explicitly instructs the upload process to do this:
https://github.com/espressif/arduino-esp32/blob/1.0.6/boards.txt#L22
```
esp32.upload.wait_for_upload_port=true
```
https://arduino.github.io/arduino-cli/dev/platform-specification/#1200-bps-bootloader-reset
>- `wait_for_upload_port` causes the upload procedure to wait for the serial port to (re)appear before and after the upload.

Since the ESP32 boards use a dedicated USB adapter chip, they do not have any sort of a serial port disappear/reappear behavior. So what happens is the upload process waits in vain for the port to appear, then finally times out and proceeds with the fallback behavior of using the port you had selected.

You might have noticed that the classic Arduino IDE does not have this behavior when uploading to the "ESP32 Dev Module". The reason is explained by the next sentence of the platform specification:
>This is only used when `use_1200bps_touch` is also set.

The classic Arduino IDE honors this, but it seems Arduino CLI (and thus Arduino IDE 2.x) does not. The "ESP32 Dev Module" board definition does not set the `use_1200bps_touch` property, so according to the specification, the `wait_for_upload_port` property should be completely ignored, yet it is not ignored by Arduino CLI.

I don't have any idea of why the ESP32 boards platform authors would have added this property to all the board definitions. From the `git blame` I see it was there from the very start. I suspect it originated from the ESP8266 boards platform, which also has `wait_for_upload_port` properties without the accompanying `use_1200bps_touch` properties in its [board definitions](https://github.com/esp8266/Arduino/blame/master/boards.txt), and thus is also subject to this delay, but I also don't find any information about the reasons for adding it there.

ullix, 2021-06-14:
Sigh. Does not make much sense to have it in the first place, but it could be "fixed" by re-adding this code to CLI. Please, do this.

PR #1319 (2021-06-14): Fix upload waiting for port without 1200bps touch
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking changes) [... cut]
closed 2021-06-14 by silvanocerza