arduino/arduino-cli #1321: [question] Sketch dependencies management

tarampampam, 2021-06-17:
What is the best way to declare sketch dependencies? For example, my sketch uses [Servo](https://github.com/arduino-libraries/Servo) library:

```cpp
#include <Arduino.h>
#include <Servo.h>

// ...
```

And when somebody wants to use this, it will be difficult for him to understand that you need to install the Servo library first. Describing all dependencies in a readme file or leaving comments with notices looks not enough clear.

It would be great if there was a command flag like `arduino-cli libs install --deps-file ./requirements.txt`:

```text
// file: requirements.txt

# https://github.com/arduino-libraries/Servo
Servo@1.1.7
```

silvanocerza, 2021-06-17:
Currently there is no way to do this and is left to the Sketch creator. We're actively discussing this to find the best way on tackling the issue. Stay tuned. :)

tarampampam, 2021-06-17:
@silvanocerza If I can help with this (I'm a go-coder) - just let me know :)

tarampampam, 2021-06-18:
If anyone wants to install sketch dependencies right now (until this is not supported by the `arduino-cli` and Arduino-IDE) - take a look at this repository: [tarampampam/arduino-deps-installer](https://github.com/tarampampam/arduino-deps-installer)

I have made a simple CLI tool for this (based on `arduino-cli` code).

closed 2021-10-05 by tarampampam

tarampampam, 2022-06-05:
Hi @r41d!

The reasons are very simple:
- The repo has no stargazers
- I hope this repo maintainers will implement the same feature independently
- I haven't the motivation for supporting the project on my side :(

JamesCohen-awin, 2023-01-19:
Follow-up note: it's possible to handle dependency management using arduino-cli's sketch project file (sketch.yaml) functionality which may have been added since this conversation.

[Build profiles documentation](../blob/master/docs/sketch-project-file.md)

laurensV, 2023-09-06:
> Follow-up note: it's possible to handle dependency management using arduino-cli's sketch project file (sketch.yaml) functionality which may have been added since this conversation.
> 
> [Build profiles documentation](../blob/master/docs/sketch-project-file.md)

Nice, this seems to be a first step in an attempt of proper library dependency managment for sketches, although not good enough yet.. We need something more similar to how libraries are managing other library dependencies in `libraries.properties` file. Example of a couple of features still missing:

- more control about library version, e.g. you need to be able to specify that you want >2.0.0 or something.
- arduino-cli commands to interact with the dependencies in sketch.json, like adding libraries to sketch.json, install all libraries defined in sketch.json etc.
- some lock file to pin library versions (and command to install libraries from lock file or update the versions in the lock file)

and many more improvements should be made, but at least we have a start..

VeloSteve, 2023-11-25:
If I could just specify something in the sketch file or some auxiliary file to open the library manager with the correct library or libraries highlighted, it would go a long way.  It doesn't have to totally automate the installation.  Version specifications would be nice, but for me "get the latest" would work fine.

I'm handing my code off to people are very smart, but complete non-programmers.  There are more of those than you might guess.  Starting off my instructions with "go to five different github repositories, two different parts of arduino.cc, and pjrc.com and install these 8 libraries" is pretty daunting.  So is "read each compiler warning and guess which things in that line might make good search terms in the library manager".  

I could really use a way to help these people out without having an hour zoom call with each one.  I know that I could distribute everything in a zip file, but each has its own licensing, so I'm not comfortable with that.

RechieKho, 2024-04-16:
I have created a sample Arduino project automated with `Makefile` that installs dependencies and board specified for reference. 
[Here is the repo](https://github.com/RechieKho/ardu).