arduino/arduino-cli #1323: Error during Upload: programmer 'atmel-ice' not available

LumitoLuma, 2021-06-17:
## Bug Report

### Current behavior

```
arduino-cli burn-bootloader -b arduino:avr:uno -P atmel-ice
```

When I try to run that command (with the Atmel-ICE connected) it yields the following output:

```
Error during Upload: programmer 'atmel-ice' not available
```

### Expected behavior

The bootloader should burn correctly with the Atmel-ICE connected

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-14T12:36:58Z
- OS and platform: Debian Testing x86_64, Linux 5.10.40

### Additional context

When trying to upload using another programmer (tried with USBasp), it succeeds.

silvanocerza, 2021-06-17:
I think it's `atmel_ice` with an underscore, could give it a try and see if it works?

silvanocerza, 2021-06-17:
Yep, it's definitely that, this is the output of `board details arduino:avr:uno`:

```
$ arduino-cli board details arduino:avr:uno
Board name:                Arduino Uno                                                 
FQBN:                      arduino:avr:uno                                             
Board version:             1.8.3                                                       

Official Arduino board:    ✔                                                           

Identification properties: VID:0x2A03 PID:0x0043                                       
                           VID:0x2341 PID:0x0243                                       
                           VID:0x2341 PID:0x0043                                       
                           VID:0x2341 PID:0x0001                                       

Package name:              arduino                                                     
Package maintainer:        Arduino                                                     
Package URL:               https://downloads.arduino.cc/packages/package_index.json    
Package website:           http://www.arduino.cc/                                      
Package online help:       http://www.arduino.cc/en/Reference/HomePage                 

Platform name:             Arduino AVR Boards                                          
Platform category:         Arduino                                                     
Platform architecture:     avr                                                         
Platform URL:              http://downloads.arduino.cc/cores/avr-1.8.3.tar.bz2         
Platform file name:        avr-1.8.3.tar.bz2                                           
Platform size (bytes):     4941548                                                     
Platform checksum:         SHA-256:de8a9b982477762d3d3e52fc2b682cdd8ff194dc3f1d46f4debdea6a01b33c14

Required tool:             arduino:avr-gcc                                                                             7.3.0-atmel3.6.1-arduino7

Required tool:             arduino:avrdude                                                                             6.3.0-arduino17          

Required tool:             arduino:arduinoOTA                                                                          1.3.0                    

Programmers:               Id                                                           Name                          
                           usbGemma                                                     Arduino Gemma                 
                           avrisp                                                       AVR ISP                       
                           arduinoasisp                                                 Arduino as ISP                
                           arduinoisporg                                                ArduinoISP.org                
                           arduinoasispatmega32u4                                       Arduino as ISP (ATmega32U4)   
                           usbasp                                                       USBasp                        
                           jtag3                                                        Atmel JTAGICE3 (JTAG mode)    
                           atmel_ice                                                    Atmel-ICE (AVR)               
                           arduinoisp                                                   ArduinoISP                    
                           buspirate                                                    BusPirate as ISP              
                           parallel                                                     Parallel Programmer           
                           stk500                                                       Atmel STK500 development board
                           avrispmkii                                                   AVRISP mkII                   
                           jtag3isp                                                     Atmel JTAGICE3 (ISP mode)     
                           usbtinyisp                                                   USBtinyISP```

LumitoLuma, 2021-06-17:
Yes, It works!

Then the line
```
Examples:
  arduino-cli burn-bootloader -b arduino:avr:uno -P atmel-ice
```
in the `arduino-cli burn-bootloader` command should be changed.

silvanocerza, 2021-06-17:
Ah, nice catch! I'll promptly fix it and then close this issue, thanks for the help. :pray:

PR #1325 (2021-06-17): [skip changelog] Fix burn-bootloader command example
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [x] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking chang [... cut]
closed 2021-06-17 by silvanocerza