arduino/arduino-cli #1329: [Nightly] Malformed board options error.

ScaredyCat, 2021-06-20:
## Bug Report
esp32:esp32 and esp8266:esp8266 cores installed via core install but an error is generated during board list and no boards are discovered. There are currently 2 network connected esp32 boards both supporting OTA which works flawlessly from the IDE.

### Current behavior

$arduino-cli board list
Error initializing instance: loading platform release esp8266:esp8266@3.0.0: loading boards: skipping loading of boards esp8266:esp8266:espduino: malformed custom board options
Port                            Type              Board Name FQBN Core
/dev/cu.Bluetooth-Incoming-Port Serial Port       Unknown
/dev/cu.SLAB_USBtoUART          Serial Port (USB) Unknown

### Expected behavior

List discoverable boards, similar to those IDE v1 port menu. Allow flashing over the network to one of those boards.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: nightly-20210620 Commit: 2173c124 Date: 2021-06-20T01:25:39Z
- OS and platform:  MacOS Mojave 10.14.6   18.7.0 Darwin Kernel Version 18.7.0: Tue Jan 12 22:04:47 PST 2021; root:xnu-4903.278.56~1/RELEASE_X86_64 x86_64

### Additional context

per1234, 2021-06-20:
Here's the change that now prevents loading any board with "malformed [custom board options](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options)":
https://github.com/arduino/arduino-cli/pull/1326

silvanocerza, 2021-06-21:
I opened a PR on the `esp8266` repo to fix this. I'll close this issue after esp8266/Arduino#8153 is merged. 
Thanks for the report @ScaredyCat.

silvanocerza, 2021-06-21:
That PR has been merged, the fix will be probably released with `esp8266@3.0.1`.
https://github.com/esp8266/Arduino/milestone/23

Guess I'll close this issue when that version is released in case other people have the same issue.

silvanocerza, 2021-07-07:
`esp8266@3.0.1` has been released on June 26, closing this PR.
https://github.com/esp8266/Arduino/releases/tag/3.0.1

closed 2021-07-07 by silvanocerza