arduino/arduino-cli #1334: core update-index results in "panic: minor version must not be prefixed with zero"

bmitov, 2021-06-23:
## Bug Report

### Current behavior

arduino-cli.exe core update-index

It returns error:
panic: minor version must not be prefixed with zero

goroutine 1 [running]:
go.bug.st/relaxed-semver.MustParse(...)
        /go/pkg/mod/go.bug.st/relaxed-semver@v0.0.0-20190922224835-391e10178d18/parser.go:19
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).loadPlatforms(0xc00068dac0, 0xc000515980, 0xc000a69ed0, 0x22, 0xc000bfb2b8)
        /home/build/arduino/cores/packagemanager/loader.go:175 +0x17f3
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectory(0xc00068dac0, 0xc00069a490, 0x0, 0x0)
        /home/build/arduino/cores/packagemanager/loader.go:120 +0x56c
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectories(0xc00068dac0, 0xc00069a4a0, 0x2, 0x2, 0x0, 0x0)
        /home/build/arduino/cores/packagemanager/loader.go:45 +0x6f
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardware(0xc00068dac0, 0xc0009f0c00, 0x0)
        /home/build/arduino/cores/packagemanager/loader.go:34 +0x46
github.com/arduino/arduino-cli/commands.createInstance(0x120dee0, 0xc00002a0e8, 0xf43c00, 0xc0009e9b01, 0xc000a682d0, 0xc000a2fb68)
        /home/build/commands/instances.go:713 +0x84c
github.com/arduino/arduino-cli/commands.Init(0x120dee0, 0xc00002a0e8, 0xc0004e3cb8, 0xc0009e9b80, 0xc000a682d0, 0xc000a2fbf8, 0x4eb3a0, 0xc000a2fbf8)
        /home/build/commands/instances.go:135 +0x5c
github.com/arduino/arduino-cli/cli/instance.getInitResponse(0x7, 0xc00068da40, 0xc000a2fd90)
        /home/build/cli/instance/instance.go:51 +0xd1
github.com/arduino/arduino-cli/cli/instance.CreateInstanceIgnorePlatformIndexErrors(...)
        /home/build/cli/instance/instance.go:35
github.com/arduino/arduino-cli/cli/core.runUpdateIndexCommand(0xc000a598c0, 0x19d14a8, 0x0, 0x0)
        /home/build/cli/core/update_index.go:45 +0x3b
github.com/spf13/cobra.(*Command).execute(0xc000a598c0, 0x19d14a8, 0x0, 0x0, 0xc000a598c0, 0x19d14a8)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:846 +0x2a4
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001298c0, 0x0, 0xc00062f8c0, 0x0)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x350
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
        /home/build/main.go:31 +0x8c

### Expected behavior

It should have updated the board index

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli_0.18.3_Windows_64bit
- OS and platform: Windows 64 bit

### Additional context

per1234, 2021-06-23:
Hi @bmitov. Please post the output you get from running this command:
```
arduino-cli config dump
```

bmitov, 2021-06-23:
C:\Arduino CLI>arduino-cli config dump
board_manager:
  additional_urls:
  - https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
  - http://download.labs.mediatek.com/package_mtk_linkitone_index.json
  - http://arduino.esp8266.com/stable/package_esp8266com_index.json
  - http://downloads.femtoduino.com/ArduinoCore-atsamd21e18a/package_atsamd21e18a-release-build_index.json
  - https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
  - https://raw.githubusercontent.com/sparkfun/Arduino_Boards/nrf5/IDE_Board_Manager/package_sparkfun_index.json
  - https://raw.githubusercontent.com/CONTROLLINO-PLC/CONTROLLINO_Library/master/Boards/package_ControllinoHardware_index.json
  - https://raw.githubusercontent.com/chipKIT32/chipKIT-core/master/package_chipkit_index.json
  - https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
  - https://macchina.cc/package_macchina_index.json
  - https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
  - https://gitee.com/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json
  - http://digistump.com/package_digistump_index.json
  - https://dl.espressif.com/dl/package_esp32_index.json
  - https://www.4dsystems.com.au/media/downloads/4Duino/package_4dsystems_index.json
  - http://apps.industrialshields.com/main/arduino/boards/package_industrialshields_index.json
  - https://raw.githubusercontent.com/feilipu/feilipu.github.io/master/package_goldilocks_index.json
  - http://dl.sipeed.com/MAIX/Maixduino/package_Maixduino_k210_index.json
  - https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json
daemon:
  port: "50051"
directories:
  data: C:\Users\mitov\AppData\Local\Arduino15
  downloads: C:\Users\mitov\AppData\Local\Arduino15\staging
  user: C:\Users\mitov\Documents\Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
sketch:
  always_export_binaries: false
telemetry:
  addr: :9090
  enabled: true

bmitov, 2021-06-23:
I cleared the entire Arduino15
After 
arduino-cli.exe config init

here is the dump:

C:\Arduino CLI>arduino-cli config dump
board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: C:\Users\mitov\AppData\Local\Arduino15
  downloads: C:\Users\mitov\AppData\Local\Arduino15\staging
  user: C:\Users\mitov\Documents\Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
sketch:
  always_export_binaries: false

arduino-cli.exe core update-index results in the same error :-(

panic: minor version must not be prefixed with zero

goroutine 1 [running]:
go.bug.st/relaxed-semver.MustParse(...)
        /go/pkg/mod/go.bug.st/relaxed-semver@v0.0.0-20190922224835-391e10178d18/parser.go:19
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).loadPlatforms(0xc00098fa80, 0xc000a4c300, 0xc000a2b2e0, 0x22, 0xc000a03d00)
        /home/build/arduino/cores/packagemanager/loader.go:175 +0x17f3
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectory(0xc00098fa80, 0xc000a2b1a0, 0x0, 0x0)
        /home/build/arduino/cores/packagemanager/loader.go:120 +0x56c
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardwareFromDirectories(0xc00098fa80, 0xc000a2b1b0, 0x2, 0x2, 0xc000a2a960, 0xc000a0bbc0)
        /home/build/arduino/cores/packagemanager/loader.go:45 +0x6f
github.com/arduino/arduino-cli/arduino/cores/packagemanager.(*PackageManager).LoadHardware(0xc00098fa80, 0x0, 0x0)
        /home/build/arduino/cores/packagemanager/loader.go:34 +0x46
github.com/arduino/arduino-cli/commands.createInstance(0x120df20, 0xc00002a0e8, 0xf43c00, 0xc000999b01, 0xc000a2a140, 0xc0009dbb68)
        /home/build/commands/instances.go:713 +0x84c
github.com/arduino/arduino-cli/commands.Init(0x120df20, 0xc00002a0e8, 0xc0009dbcb8, 0xc000999b20, 0xc000a2a140, 0xc0009dbbf8, 0x4eb3a0, 0xc0009dbbf8)
        /home/build/commands/instances.go:135 +0x5c
github.com/arduino/arduino-cli/cli/instance.getInitResponse(0x7, 0xc00098fa00, 0xc0009dbd90)
        /home/build/cli/instance/instance.go:51 +0xd1
github.com/arduino/arduino-cli/cli/instance.CreateInstanceIgnorePlatformIndexErrors(...)
        /home/build/cli/instance/instance.go:35
github.com/arduino/arduino-cli/cli/core.runUpdateIndexCommand(0xc000a158c0, 0x19d14a8, 0x0, 0x0)
        /home/build/cli/core/update_index.go:45 +0x3b
github.com/spf13/cobra.(*Command).execute(0xc000a158c0, 0x19d14a8, 0x0, 0x0, 0xc000a158c0, 0x19d14a8)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:846 +0x2a4
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001298c0, 0x0, 0xc000987c20, 0x0)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:950 +0x350
github.com/spf13/cobra.(*Command).Execute(...)
        /go/pkg/mod/github.com/spf13/cobra@v1.0.1-0.20200710201246-675ae5f5a98c/command.go:887
main.main()
        /home/build/main.go:31 +0x8c

Tested on a clean VM It appears to work fine. Older versions on the main system work fine. How to make the new version to work on my main system? What could be causing this?

bmitov, 2021-06-24:
Ok, I have been running with:
go run main.go core update-index
Installed go and everything. I got it to work to a point. Now I get:
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
Updating index: package_damellis_attiny_index.json downloaded
Updating index: package_mtk_linkitone_index.json downloaded
Updating index: package_esp8266com_index.json downloaded
Updating index: package_atsamd21e18a-release-build_index.json downloaded
Updating index: package_adafruit_index.json downloaded
Updating index: package_sparkfun_index.json downloaded
Updating index: package_ControllinoHardware_index.json downloaded
Updating index: package_chipkit_index.json downloaded
Updating index: package_stm_index.json downloaded
Updating index: package_macchina_index.json downloaded
Updating index: package_nRF5_boards_index.json downloaded
Updating index: package_esp32_index.json downloaded
Updating index: package_digistump_index.json downloaded
Updating index: package_esp32_index.json downloaded
Updating index: package_4dsystems_index.json downloaded
Updating index: package_industrialshields_index.json downloaded
Updating index: package_goldilocks_index.json downloaded
Updating index: package_Maixduino_k210_index.json downloaded
Updating index: package_seeeduino_boards_index.json downloaded
Error updating index: invalid package index in https://files.seeedstudio.com/arduino/package_seeeduino_boards_index.json: json: invalid number literal, trying to unmarshal "\"104103008 \"" into Number
exit status 1

The "\"104103008 \"" has extra space. This does not bother the Arduino IDE, and it works with older versions of Arduino-CLI.
Is there a way to modify the code to trim the extra space in numeric values?

bmitov, 2021-06-24:
I fixed it by changing:
type indexToolReleaseFlavour struct {
	OS              string      `json:"host,required"`
	URL             string      `json:"url,required"`
	ArchiveFileName string      `json:"archiveFileName,required"`
	Size            json.Number `json:"size,required,string"`
	Checksum        string      `json:"checksum,required"`
}

I added ,string filter to the Size
Can you please add in the repository, or if you give me permission I can push the fix with GIT

per1234, 2021-06-24:
>The ""104103008 "" has extra space.

I saw that also when I did a quick test with your list of Boards Manager URLs. However, it doesn't seem to be related to the original error you reported:
```
panic: minor version must not be prefixed with zero
```
After trimming the string, does that error issue go away?

>if you give me permission I can push the fix with GIT

Not even Arduino's developers are allowed to push fixes directly. All changes are done via [pull requests](https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests). Everyone is very much welcome to submit pull requests. They don't require any special permissions.

bmitov, 2021-06-24:
I have discovered what was causing the other error. It was a platform.txt file that had :
version=1.00.

instead of

version=1.0.0

It was outdated file, and they had already fixed this.

I also created a pull request for the seeeduino fix.

Also learned how to debug GO in VSCode, so at least got some benefit from it :-D

closed 2022-10-12 by cmaglie

per1234, 2022-10-12:
I verified the original issue is fixed. If you set the `version` property in the `platform.txt` file of an installed platform to an invalid semver format like:

```text
01.00.00
```

With Arduino CLI 0.18.3, `core update-index`, `core list`, `compile` (I didn't check others) panic with the output provided in the original issue report.

Using the latest build of Arduino CLI (https://github.com/arduino/arduino-cli/commit/589cbccec2b1dc4d7a53c916e491156e8c617c5c), `core update-index` is successful and the other two commands display a user friendly error message of this form:

```text
Error initializing instance: Error loading hardware platform: Invalid version: C:\Users\per\Documents\Arduino\hardware\arduino-dev\avr\platform.txt: major version must not be prefixed with zero
```