arduino/arduino-cli #1336: Controllino board not supported

gbataille, 2021-06-24:
## Bug Report

### Current behavior

```bash
gbataille ~/Doc…DC/BiU_Corn/plc [1] $ arduino-cli compile --fqbn CONTROLLINO_Boards:avr:controllino_mega controllino --verbose
Using board 'controllino_mega' from platform in folder: /Users/gbataille/Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/3.1.0
Using core 'arduino' from platform in folder: /Users/gbataille/Library/Arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -I/Users/gbataille/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/Users/gbataille/Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/3.1.0/variants/Controllino_mega /var/folders/cx/327j2f6j2j19z_tyjz5gqmtc0000gp/T/arduino-sketch-3F38DC76432B994275324F06CC5233C0/sketch/controllino.ino.cpp -o /dev/null

Error during build: fork/exec /Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++: bad CPU type in executable
```

### Expected behavior

Compilation should work

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-21T17:11:27Z`
- OS and platform: MacOS 11.4

### Additional context

Hi everyone,

I understand that you are not responsible for the support of all the boards.
That being said, I have a Controllino board (that is actually an enhanced atmega). It works well with Arduino IDE 1.8.
However, with both arduino IDE 2.0 and arduino-cli, I get the above error

I have raised the issue with the controllino guys. They have not worked on compatibility yet but they are surprised about this error, thinking that the compilation command looks good.
See github.com/CONTROLLINO-PLC/CONTROLLINO_Library/issues/46

I suspect this has something to do with the configuration file they provide for the board.
Would you be able to maybe quickly see what's the issue?
I'm actually hoping that you'll have the answer or a hint just by looking at the trace

Thanks

per1234, 2021-06-24:
Hi @gbataille the problem is shown here:
```
Error during build: fork/exec /Users/gbataille/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++: bad CPU type in executable
```
Note the avr-gcc version: 4.9.2-atmel3.5.3-arduino2. This is a very outdated version. Support for 32-bit applications was dropped from macOS in the 10.15 release. Arduino rebuild the macOS toolchains to 64-bit in response (https://github.com/arduino/Arduino/issues/8974), but this version of avr-gcc dates from before that time.

So the Controllino boards platform maintainers will simply need to make a new release that uses the modern AVR toolchain. Information about that here:
https://arduino.github.io/arduino-cli/dev/package_index_json-specification/#platforms-definitions

closed 2021-06-24 by per1234

gbataille, 2021-06-24:
Awesome, thanks. I guessed there was some instant knowledge that could be had by just asking the right person :)

I'll push the information to them.

Thanks for your super quick answer.

pmmarquez, 2024-01-19:
Hello this is Pedro from Controllino,

Also we have observed the similar issue with avrdude version, we are currently working on it but until the board support gets updated this can be a solution:

- Close Arduino IDE
- Go to /Users/***USER***/Library/Arduino15/packages/arduino/tools/avrdude and copy the name of the file of the latest version of avrdude for example "6.3.0-arduino17" and delete the “6.3.0-arduino8” folder.
To be sure in the compilation and upload log in the Arduino IDE you can see the exact version that UNO and MEGA are using, this one will work:
[image]
- Open /Users/***USER***Library/Arduino15/packages/CONTROLLINO_Boards/hardware/avr/3.1.0/installed.json and change "version": "6.3.0-arduino8" to the copied version for example "version": "6.3.0-arduino17"
- Open /Users/***USER***/Library/Arduino15/package_ControllinoHardware_index.json and change "version": "6.3.0-arduino8" to the copied version for example "version": "6.3.0-arduino17"
- Open again the Arduino IDE

This should force the Arduino IDE to use a different version of avrdude, similar solution can be applied to any package or tool version just by seen which one is used while compiling or uploading to an Arduino UNO or MEGA.

Best Regards 👍