arduino/arduino-cli #1338: .h file are preprocessed

fpistm, 2021-06-28:
**Describe the bug**
STM32 core allows user to define some value by adding a build_opt.h file at sketch level.
This file is not a standard header file as it is used by the gcc command line with `@file` option.

https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

To achieve this a simple recipe hooks prebuild is used. When no file exists at sketch level it creates an empty one.
It works fine until Arduino 1.8.13. Since 1.8.15 it is failed.
`.h` file extension has been used because Arduino 1.8.x does not allow to create file without a specific extension (ino, c, cpp, h, hpp).

I guess it is linked to https://github.com/arduino/arduino-cli/pull/1224

For example a build_opt.h with:
`-DHAL_CRC_MODULE_ENABLED`

is converted to:
```
#line 1 "c:\\Users\\<username>\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2021410-37444-aykylf.uyhtw\\sketch_may10a\\build_opt.h"
-DHAL_CRC_MODULE_ENABLED
```

Below the build result due to the `#line` addition:
Using command:
```console
$ arduino-cli compile -b STMicroelectronics:stm32:Nucleo_144:pnum=NUCLEO_H743ZI2 C:\\STM32\\arduino\\arduino-1.8.15\\portable\\sketchbook\\blink_minimal\\blink_minimal.ino
arm-none-eabi-g++: error: #line: No such file or directory
arm-none-eabi-g++: error: 1: No such file or directory

Error during build: exit status 1
```

**To Reproduce**
Steps to reproduce the behaviour:
1. Install STM32 core version 2.0.0  (https://github.com/stm32duino/BoardManagerFiles/raw/master/package_stmicroelectronics_index.json)
2. Add a `build_opt.h` file containing the line `-DHAL_CRC_MODULE_ENABLED` to an example
3. Build the example:
`arduino-cli compile -b STMicroelectronics:stm32:Nucleo_144:pnum=NUCLEO_H743ZI2 C:\\STM32\\arduino\\arduino-1.8.15\\portable\\sketchbook\\blink_minimal\\blink_minimal.ino
`
**Expected behavior**
Do not preprocess .h file and able to build.

**Desktop (please complete the following information):**
- CLI version (output of `arduino-cli version`):  arduino-cli.exe alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-14T12:36:58Z
- OS and platform: all

**Additional context**

Issue originally reported by an user here:
https://github.com/stm32duino/Arduino_Core_STM32/issues/1387
https://github.com/arduino/arduino-ide/issues/363

Maybe a fix would be to add an extra file extension to avoid to parse it. Ex: .opt or simply .txt.

matthijskooijman, 2021-07-06:
@fpistm Would it not be better to fix this by using a different extension that .h? Since `build_opt.h` does not actually contain C/C++ code, but compiler commandline options, so `.h` does not seem appropriate? Or is it needed to ensure that the file is copied into the temp directory when compiling a file?

fpistm, 2021-07-06:
@matthijskooijman 
Yes we agree,  as mentioned:

>  `.h` file extension has been used because Arduino 1.8.x does not allow to create file without a specific extension (ino, c, cpp, h, hpp).

And this is exactly what I've suggested at the end of my OP:

> Maybe a fix would be to add an extra file extension to avoid to parse it. Ex: .opt or simply .txt.

For me ".opt" is fine and more explicit: build.opt. And is inline with gcc documention: `Read command-line options`
https://gcc.gnu.org/onlinedocs/gcc/Overall-Options.html

About the copy I don't yet think about it but it can always be done by the prebuild recipe.

silvanocerza, 2021-07-06:
Just chiming in to say that we're not ignoring this. We're focusing on developing the [Pluggable Discovery][pluggable-discovery] and it's taking most of our time, I want to take a look at this as soon as we're done with the Pluggable. No ETA for now. 

[pluggable-discovery]: https://github.com/arduino/tooling-rfcs/blob/main/RFCs/0002-pluggable-discovery.md