arduino/arduino-cli #1339: SAM is not available on aarch64

Kakcalu13, 2021-06-30:
## Bug Report
[image]

### Current behavior

```
ubuntu@ubuntu:~/arduino-cli$ arduino-cli core install arduino:sam
Downloading packages...
Error during install: tool arduino:arm-none-eabi-gcc@4.8.3-2014q1 not available for the current OS
ubuntu@ubuntu:~/arduino-cli$ 
```
I need it for arduino DUE via RPI.

### Expected behavior

It should be added just like this laptop:
[image]

### Environment

- CLI version (output of `arduino-cli version`):
- OS and platform:
[image]
This is raspberry pi 3 model b+

### Additional context
AVR and SAMD were downloaded without any issue, but not SAM. Do you know why did this happen? Is SAM considered as 3rd party?

per1234, 2021-06-30:
Hi @Kakcalu13. Thanks for your interest in this open source project.
>Do you know why did this happen?

It's because the Arduino SAM Boards platform is using old tools versions from before there was any need to build them for 64 bit ARM Linux.

This was previously discussed at https://github.com/arduino/Arduino/issues/10579. If you check the links off of that, you might discover a workaround.

>Is SAM considered as 3rd party?

No, it's one of the official Arduino boards platforms.

Since this issue is not related to Arduino CLI, I'll go ahead and close this as off topic for this repository.

closed 2021-06-30 by per1234

mwimmer88, 2025-01-16:
here i post a workaround using an inoffical package provieded by per1234

https://github.com/arduino/Arduino/issues/10579#issuecomment-2595231603