arduino/arduino-cli #1343: Warning: tool 'arduino:openocd@0.11.0-arduino2' is not installed. It might not be available for your OS.

rmlearney-digicatapult, 2021-07-01:
## Bug Report

Warning: tool 'arduino:openocd@0.11.0-arduino2' is not installed. It might not be available for your OS.

### Current behavior

Compiling and uploading from VSCODE 1.57.1 with Arduino Extension 0.4.3 on MacOS 10.15.7

Getting the error message above just before 1200bps touch

### Expected behavior

Should this tool be installed?

### Environment

- CLI version (output of `arduino-cli version`): 0.18.3 Commit: d710b642
- OS and platform: MacOS 10.15.7

per1234, 2021-07-01:
>Arduino Extension 0.4.3

This 3rd party extension is not supported here. Does the issue occur after you install and upload to your board using Arduino CLI directly from the command line?

RMLEARNEY, 2021-07-12:
Never tried uploading from the command line...not sure how to

RMLEARNEY, 2021-08-14:
Fixed by running `arduino-cli update` and `arduno-cli upgrade` from the command line - seemed to identify the missing package and pull it down

rmlearney-digicatapult, 2021-08-14:
Fixed by running `arduino-cli update` and `arduno-cli upgrade` from the command line - seemed to identify the missing package and pull it down

closed 2021-08-14 by rmlearney-digicatapult