arduino/arduino-cli #1347: Running command Together uses same port

TravisDavis-ops, 2021-07-04:
## Bug Report

### Current behavior
arduino-cli upload --port /dev/ttyACM1 -b arduino:mbed_nano:nano33ble ./writer \
arduino-cli upload --port /dev/ttyUSB0 --fqbn esp8266:esp8266:generic ./reader

command 2 use command 1's port

### Expected behavior
upload board 1 
upload board 2

### Environment

- CLI version (output of `arduino-cli version
arduino-cli alpha Version: 0.18.3-arch Commit: d710b642ef7992a678053e9d68996c02f5863721 Date:
-  OS and Platform
uname -a Linux ArchBox1 5.12.9-zen1-1-zen #1 ZEN SMP PREEMPT Thu, 03 Jun 2021 11:36:07 +0000 x86_64 GNU/Linux

### Additional context

# Full Output 
Device       : nRF52840-QIAA
Version      : Arduino Bootloader (SAM-BA extended) 2.0 [Arduino:IKXYZ]
Address      : 0x0
Pages        : 256
Page Size    : 4096 bytes
Total Size   : 1024KB
Planes       : 1
Lock Regions : 0
Locked       : none
Security     : false
Erase flash

Done in 0.001 seconds
Write 85288 bytes to flash (21 pages)
[==============================] 100% (21/21 pages)
Done in 3.308 seconds
esptool.py v3.0
Serial port /dev/ttyACM1 //wrong port
Connecting.....^C
Traceback (most recent call last):
... Suff about not begin an esp8266
KeyboardInterrupt

silvanocerza, 2021-07-05:
Interesting. Does this happen only when one of the boards is an `esp8266`? :thinking:

matthijskooijman, 2021-07-05:
It looks a bit like esptool.py is using the wrong port somehow, I wonder how that configures the port to use. Maybe it uses some global config file instead of a commandline option for some reason?

@TravisDavis-ops, could you post verbose output of both commands? i.e. add `-v` to both commands?

Also, are you running both commands at the same time? Or does the problem also occur when you run them after each other?

TravisDavis-ops, 2021-07-05:
https://gist.github.com/TravisDavis-ops/89d65a51bde571e1f359ec97ef0957c0

silvanocerza, 2021-07-06:
Does it happen with `bash` or `zsh` too?

matthijskooijman, 2021-07-06:
I'm still not sure I understand the problem entirely. In your gist, I see a reader and writer log, I suspect this is the output you redirected to a file? What about your arduino-cli.log, is that log output from both arduino-cli runs? Looking at the output, I see a couple (more than two) of `arduino-cli version 0.18.3-arch` lines that, I suspect, start a new arduino-cli run?

Maybe you can show the *exact* command you ran, including any redirections or other info that helps us understand how your log output was made?

In any case, IIUC, the `Upload port found on /dev/ttyACM0` message does come from arduino-cli, so it's not just esptool.py that is confused about the port, it's also arduino-cli itself I believe?

TravisDavis-ops, 2021-07-06:
made sure to clean up before runing 
yes i am pipeing thing i included that command as well