arduino/arduino-cli #1350: Compiling fails when there's a build_opt.h present in the folder

RalphSommerer, 2021-07-06:
When compiling an Arduino project for the Arduino_Core_STM32 that contains a build_opt.h header file (to add further compile options), the following error is produced:

```
Arduino: 1.8.15 (Windows 10), TD: 1.54, Board: "Generic STM32F1 series, BluePill F103CB (or C8 with 128k), HID Bootloader 2.2, Enabled (generic 'Serial'), CDC (generic 'Serial' supersede U(S)ART), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

arm-none-eabi-g++: error: #line: No such file or directory

arm-none-eabi-g++: error: 1: No such file or directory

exit status 1

Error compiling for board Generic STM32F1 series.
```

It used to work with the previous version of the Arduino IDE (1.8.13 or whatever), although it did fail with the same error when compiling using the Arduino-Cli.

I presume some sort of preprocessor failure (hence the error on the "#line" macro).

**To Reproduce**

Add a "build_opt.h" file to the folder containing the main sketch, then compile.

Contents of the "build_opt.h":

```
-DHAL_TIM_MODULE_ONLY
-DHAL_EXTI_MODULE_DISABLED

```

**Expected behavior**
Compilation succeeds (unless there are errors unrelated to this issue, obviously...)

**Desktop**
 - OS: Windows 10
 - Arduino IDE version: 1.8.15
 - STM32 core version: 2.0.0
 - Tools menu settings: see above error output
 - Upload method: HID Bootloader 2.2

**Board**
 - Bluepill F103C8 (with 128k)

**Additional context**
Again, it compiled just yesterday without error, before I upgraded to the latest version (1.8.15) of the Arduino IDE

per1234, 2021-07-06:
Hi @RalphSommerer. Thanks for taking the time to submit an issue.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-cli/issues/1338.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-07-06 by per1234

RalphSommerer, 2021-07-06:
I searched but didn't find....

Thanks!