arduino/arduino-cli #1364: Build failure for Linux 64

4ntoine, 2021-07-23:
## Bug Report

### Current behavior

task dist:Linux_32bit &> build_linux_64bit.log

I'm trying to compile it for Linux 64bit.
The compilation fails with error code 1 (log file attached below).

### Expected behavior

The source code is compiled into the binaries with no errors.

### Environment

- CLI version (output of `arduino-cli version`):
0.18.3

- OS and platform:
macos mojave (10.14.6)

### Additional context

Here is the build log:
[build_linux_64bit.log](https://github.com/arduino/arduino-cli/files/6869185/build_linux_64bit.log)

PS. Let me know if i can help with anything.

silvanocerza, 2021-07-23:
Hi @4ntoine, sorry but the log file seems completely mangled after line 83, this is what I see:

```
go: downloading github.com/shurcooL/sanitized_anchor_name v1.0.0
^_<8B>^H^@^U<C8><FA>`^@^C<EC><BD>^KxSE^V8~<93>&<90>b<EA>^MP0`<90>T<A3><B6>X<A0>Ѫ<8D><82><A6>P<B0><B0>U<8B>^T(X^U^UY<C0>*<A8>    <A0>VhIB{<BD>ܵ*<88><AB><EE>.<BB>>V^P<DF>ny<88>n<DA>^@<E5>Ⴧ"^OQPם^PE<8<81><95><F2><EE><FF><9C>3<F7><DE>$m<8A><EE><EF><FB><FD><FF><DF><F7><FF><BE><ED><D7>$<F7><CE>=s<E6>̙3gΜ93<F7><CE>^G'<F9><A7><DE>?<BD><FF>ݕS<85><FF><B7><FE><F2><F2><F2><AE><BE><F2>J'<FD>^<C5>^?<F3>.<<CF><C7>_<F7><95><97>_uy<DE>^UW_<EE>t<E7><E3><B3><FC>||<90><E7>ο<<FF>
<C1><99><F7><FF>^ZE     ^?<FE><87>|w<A4><DC><F9><D0>}S^_<BC>^?<FA>̎<E0>^@l<F2><E4><B3><E0>ɣ?<A7><FE><FB><FF><93><BF><B9><C3>J<86>^F<FD><DE>(<A4> x<B7><EF><F6>.<96>|<F8><F5><A9><8F><F2>^E<A7><D0>E<C<EA><AF><D9>^X<87>^K<D7>>n      w<9E>/<E4>^Y<DB><D3>>C<C5>G<8F><90>w<C0>ÕW_<A3><C3>9<D5><DF>^V^@JW<E1>V6<AF><B5><AC><BC>!,X<E0><FA><9C>^D|x^?*^A.^Lpa<80><B3>^Y<DA><C3>u^A<80><D6><D6><D6><E9>^H<B7>o<C3>Z˾<E2><B0>pS^B><A3><FA>9^O<BE>Ψpލk-<DE>^Qa<A1>^^<F3><AB><CF>5<DE>m<82>2<CC>j<DA>><80><DB>^Gp<E1>6<F4><E1>g<A5><81>ˢ<F6><B7>ddX<B8>$<81>/<9A><9C><FD><A4><C2>^Q<BE><8F>^@_IX(O^A<F7>o<B5>nDߧ@<DF<DF><CD>aa<86><91>ӧ<C9>5<B6><C9><EC>^D><87><B7>^B_F<85><85><8A><BA>8><8D><CF><F3>^R<E0>^\^S<D6>Y^\<BB><C3>B<F6>yq8<AD><BE>^K^R<E0><8A>^^Xg):^T^V^R<FF>4<FA><9E>I<80><9B>^Hp^S^A<AE><DF>mw<B7>+w<88>
G<BC><C1>>^B}<C1>)<C4><FF>4<B9><FA><DC>^P<87>s^B<9C>^S<E0>^V&<C0>i|<FE><CA><C0>e^O<E1><A8>O^A^\<F6><A7>.        p(^G[^R<F0><95>o<CD>N/^_<E5>2<98>R<E0><EB><9C>@<DF>D<80><9B>^Hp<E1>^W<CC><ED><EA><D1<D1>-^A<CE>y<FF><80>t<E7><C1>^\C<F8>{<A1>^]<9C>]-^_<E1><F6><95>^ML<DF><F7>Y?<C3>{C^M<ED><E0><FA>$<C0>-y6/}<89>=<D7>P<EE>m<8F><AF>T<83>K <BE>m=<F0><EF><9F>B<82>.ۗ<97>.<EC><EB>g<E8>eh^Ow<B1>!^N<D7>k^^?^z<AF><FD><FD>^L<A3>~<EC><D1>^NnO^B<BE>i<9F>^LO<9F><F6><89><D7><F0><C2>7<A6>vpC^R<F0>uz<FB>w<E9><9D><DE>^^f<D8><F8>b<FB>r<EF>N<80><FB><B4><AC>$<FD>Ӳᆿ<86><8C><ED><E0>| p3vߔ>cw<B1><A1>_B}<B5>g<BF>$\?^Gp<CF>^A<DC><F2>-^W<B6>×<91><80>o<F8>k^O<A4>^O^?<ED>^<C3><EE><B5>絃[<9A>^@<E7><9C>!<A5>;g^D^Lu<E7>tn^G<D7>d<88><CB>=<8D>9"^_o<DA><D2>7Vಅp4<F6><88>|<DC>i<8B><EF><AF>*<8C><FE>W<FD><B8>a      <8C><F2><FB>I<95>EW^O<AC>|<F8><F7>7<DC>r<CF>^U<9E><A2>|<FF>]W<DE>_>~<94><EF><FE><BC><81><C3><C6>^L<F5><DF>_~<E3><BD><F9>W_u<D7><E5>^O?RP0vD<D9><C0>ʑ<F7>^}<FF>}ySKfܛ<E7><BF><F2><FE><E9><A3>G<8E>^E<AB>u`<E5>Ի^FVN<EA>_9<F5>~<FF><EC>^A^OM^_p<B9>0f<DC><D8><D1>l%^TZ^S<B9><B9><D0><FC><E0><90>}<AE><<F6>y<A3>Y<90><8A>\V<B6>^\曁^_L<CD>Dd<EB>L{<EB><CC><CC>֙6<A9><B9>u<A6><B5>u<A6><A5>u<A6><A9><F5>*<F6>,¬uJ<EB><E4><EB><D9>^K<90><F1>2<C8>(Uvr^D<D6>:<A4>c<B7><DE>~<C7>m^Qe<A6><A0><AC><B9>^TQPy<C5>T<DE><D7>^,<CF>
<E5>9<D9>=<90>^M<8A>A<BC>j^Q<89>^E\<AD>^U<C0>^F^C\<CE>1<8E><DC>J<A8>e<87>+<F2>Q<C7>^?T<DE>N<98>}<D6>D<A6>{<E3><F5><FB>%<AC><D6>!A<AB><9C>^U<8B>W<8B><FC>^L<D2><E5><EB><A1><C4><CD>a<AA><90>^Cʴ^E<D6<D6><E6><AA>^U<92><8E><AA><F5>^Y<CD>^^%<E4><DD>^Hy<A6>^L<90><AD>y앣H<B0><FD><D6>H=<95>_N ?]^_<AF><AF>?<8C><F5><C5>^B<AF>b^R^@CAw<85>ժ<99><80>o<9C>k<BE>K8<BF><EC><94><FF>̈́<FC>W"<FD>I<8D>p^Q<C7><E2>а
<D8>4,<C9><FC>^@l<9B><CE> <B6>; <9B>I~^Wk<A1>^D<AE><9B><FE>lk<EB>U<D3>2<A6>^YJ<99>^EP)E.^W<B0><B8><B4><D5><E7><B2>^Cv3$<95>O^@<80>.<AC>^K\<B2>^^<F0>E<F8><90><BC><87>   <A1><8D>^P:<94>w<EB><B7>v^V<<94><F7>_<85><EF><B1><EC><F4>/<C0>Ƨ^W<C0>u`<AD>-<B4><CD>^?^<E0>^G<CB><D8>q<EC>(&^G<B4><E4><B0>/<8A><F5>z<BF><8E><B2>^\<F8>^Eygs<87>C<AD><BE>^<CD><D0>FH <^G<8A>쥭_ <87><E1>1<FC>C<ED>"<F5>P<FC><89><<D3>X<FC><93><D7>a<F1>6^<FC>X^V^D,J^@<89>h<A6>r<D0>^V<97>L.<F6>^B<A4>K<F9>.z^^X<9B>y<EB><ED>^Q<E5><BD>K<88>^E<BC>=^A<DF>2<C2>W<98><8C>/^_).r9!<AF>^T@Bِ8<86>^FB<80><D3>=^A<F9>1<95>^P^H<84><C0><AA>!8<F2>^_@@U<C4><E2>ͼ<92>$MI<F9><B1><BD>?<F4>^N^Q^Ev^QaY=<98>s<B5><D2>e<A9>Gy<80><F6>^X<94><F8>a{<FE>^S<E2>L<94><9A>}<D7>^D<8E><9D><E3>ϖ|.+<F0><CD><C2><F9>f<99>&<8C><A9>YO^X<C3>^\<B4><E6><8C>w<BA><D7<D7><E0><FF>GR<F7><C4>^F(^B<94>yl<D9>^?T<D2>t1lH<E0><CF>h&<9F>B<BA>.$<BA><EC>@<82>^K?<D3>^Dj^^6^X$<95><A8><F1><9B>I<FE>^Am)<9B>B^X<AD><A5>J`<DA><CD>^?l-<95>^C<D4><DD><F2><D8>DHg<9B><8F><A8><92><F4<F4><91>&O@<87><8D>]L<A5><AC>^Y<C4>E<BE>@<9E>a)e^<80><97>^F<B9>Z<9F>vn<E4>\<^M<AD>J<A2>^R<E7>"<B0><FF><C0>I<CC>ZCY<BD>
<C8>/ ,^]+^E0^W<B3><FE>^G^E^Y<88><B5><95><8E>e"<DC>(A<97>^C&s<D0><C5><ED>r<99><AB>T2"<E2>N<F8> <90><B7><91><E4>S^V@Bm<C0>I<A8>h^Y}<B6><90>^@<E6>C<D3><F3><FE>=<9B>J^T<A1>Ě^_ʁ&y<A1><AB>^R~<A0><EC>^R<A9><
```

Sadly I can't help much without the log. :confused:

4ntoine, 2021-07-23:
Here is the file for 64 bit:
> task dist:Linux_64bit &> build_linux_64bit.log

[build_linux_64bit.log](https://github.com/arduino/arduino-cli/files/6869240/build_linux_64bit.log)

4ntoine, 2021-07-23:
@silvanocerza that's right, that's the output i have. Anything i should do instead?

4ntoine, 2021-07-23:
```
> task --version
Task version: v3.6.0 (h1:RYrzgAOwuyvE6fR3xMaBzQseYKuE7xwQC4SRrpG4TRU=)
```

silvanocerza, 2021-07-23:
I just tried building on OS X and it's failing for me too but the output at least is readable.

```
go: downloading gopkg.in/warnings.v0 v0.1.2
# google.golang.org/protobuf/encoding/protojson
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Error: failed building for linux/amd64: exit status 2
failed building for linux/amd64: exit status 2
task: Failed to run task "dist:Linux_64bit": exit status 1
```

Am on an M1 with Big Sur 11.4, so it's possible that it's not the same issue you're having. 🤔

silvanocerza, 2021-07-23:
Ah! This might be a Taskfile issue I think, running `task dist:Linux_64bit` prints the output correctly in the terminal but piping it to a file mangles it. 
Can you try running it without piping?

4ntoine, 2021-07-23:
That's pretty much the same:

```
> task dist:Linux_64bit
task: Ädist:Linux_64bitÅ docker run -v épwdé/..:/home/build -w /home/build ö
-e CGO_ENABLED=1 ö
docker.elastic.co/beats-dev/golang-crossbuild:1.14.7-main ö
--build-cmd "go build -o dist/arduino-cli_linux_amd64/arduino-cli -ldflags ' -X github.com/arduino/arduino-cli/version.versionString=0.18.3 -X github.com/arduino/arduino-cli/version.commit=d710b642 -X github.com/arduino/arduino-cli/version.date=2021-07-23T14:21:39Z '
" ö
-p "linux/amd64"

tar cz -C arduino-cli_linux_amd64 arduino-cli -C ../.. LICENSE.txt  -f arduino-cli_0.18.3_Linux_64bit.tar.gz
sha256sum arduino-cli_0.18.3_Linux_64bit.tar.gz >> 0.18.3-checksums.txt

>> Building using: cmd='go build -o dist/arduino-cli_linux_amd64/arduino-cli -ldflags ' -X github.com/arduino/arduino-cli/version.versionString=0.18.3 -X github.com/arduino/arduino-cli/version.commit=d710b642 -X github.com/arduino/arduino-cli/version.date=2021-07-23T14:21:39Z '
', env=ÄCC=gcc, CXX=g++, GOARCH=amd64, GOARM=, GOOS=linux, PLATFORM_ID=linux-amd64Å
go: downloading google.golang.org/grpc v1.27.0
go: downloading github.com/spf13/cobra v1.0.1-0.20200710201246-675ae5f5a98c
go: downloading github.com/gofrs/uuid v3.2.0+incompatible
go: downloading github.com/fatih/color v1.7.0
go: downloading github.com/arduino/go-paths-helper v1.4.0
go: downloading github.com/arduino/go-win32-utils v0.0.0-20180330194947-ed041402e83b
go: downloading github.com/sirupsen/logrus v1.4.2
go: downloading github.com/mattn/go-colorable v0.1.2
go: downloading github.com/spf13/jwalterweatherman v1.0.0
go: downloading github.com/cmaglie/go.rice v1.0.3
go: downloading github.com/segmentio/stats/v4 v4.5.3
go: downloading golang.org/x/text v0.3.2
go: downloading github.com/leonelquinteros/gotext v1.4.0
go: downloading go.bug.st/relaxed-semver v0.0.0-20190922224835-391e10178d18
go: downloading golang.org/x/net v0.0.0-20200324143707-d3edc9973b7e
go: downloading google.golang.org/protobuf v1.25.0
go: downloading gopkg.in/src-d/go-git.v4 v4.13.1
go: downloading github.com/pkg/errors v0.9.1
go: downloading go.bug.st/downloader/v2 v2.1.1
go: downloading github.com/arduino/go-properties-orderedmap v1.3.0
go: downloading github.com/golang/protobuf v1.4.2
go: downloading github.com/arduino/board-discovery v0.0.0-20180823133458-1ba29327fb0c
go: downloading github.com/marcinbor85/gohex v0.0.0-20210308104911-55fb1c624d84
go: downloading github.com/cmaglie/pb v1.0.27
go: downloading github.com/arduino/go-timeutils v0.0.0-20171220113728-d1dd9e313b1b
go: downloading github.com/codeclysm/extract/v3 v3.0.2
go: downloading golang.org/x/sys v0.0.0-20200909081042-eff7692f9009
go: downloading github.com/daaku/go.zipexe v1.0.0
go: downloading github.com/codeclysm/cc v1.2.2
go: downloading github.com/juju/errors v0.0.0-20181118221551-089d3ea4e4d5
go: downloading github.com/mattn/go-isatty v0.0.8
go: downloading golang.org/x/crypto v0.0.0-20200406173513-056763e48d71
go: downloading github.com/spf13/pflag v1.0.3
go: downloading github.com/emirpasic/gods v1.12.0
go: downloading gopkg.in/src-d/go-billy.v4 v4.3.2
go: downloading go.bug.st/serial.v1 v0.0.0-20180827123349-5f7892a7bb45
go: downloading github.com/src-d/gcfg v1.4.0
go: downloading google.golang.org/genproto v0.0.0-20200526211855-cb27e3aa2013
go: downloading github.com/jbenet/go-context v0.0.0-20150711004518-d14ea06fba99
go: downloading github.com/oleksandr/bonjour v0.0.0-20160508152359-5dcf00d8b228
go: downloading github.com/kevinburke/ssh_config v0.0.0-20190725054713-01f96b0aa0cd
go: downloading github.com/h2non/filetype v1.0.8
go: downloading github.com/rifflock/lfshook v0.0.0-20180920164130-b9218ef580f5
go: downloading github.com/sergi/go-diff v1.1.0
go: downloading github.com/xanzy/ssh-agent v0.2.1
go: downloading go.bug.st/serial v1.1.2
go: downloading github.com/spf13/viper v1.6.2
go: downloading github.com/fluxio/multierror v0.0.0-20160419044231-9c68d39025e5
go: downloading github.com/mitchellh/go-homedir v1.1.0
go: downloading github.com/cpuguy83/go-md2man/v2 v2.0.0
go: downloading github.com/segmentio/fasthash v0.0.0-20180216231524-a72b379d632e
go: downloading gopkg.in/warnings.v0 v0.1.2
go: downloading github.com/mattn/go-runewidth v0.0.9
go: downloading github.com/miekg/dns v1.0.5
[... 3 lines cut ...]
go: downloading github.com/creack/goselect v0.1.1
go: downloading gopkg.in/yaml.v2 v2.3.0
go: downloading github.com/magiconair/properties v1.8.1
go: downloading github.com/hashicorp/hcl v1.0.0
go: downloading github.com/pmylund/sortutil v0.0.0-20120526081524-abeda66eb583
go: downloading github.com/fsnotify/fsnotify v1.4.7
go: downloading gopkg.in/ini.v1 v1.51.0
go: downloading github.com/spf13/afero v1.1.2
go: downloading github.com/subosito/gotenv v1.2.0
go: downloading github.com/pelletier/go-toml v1.2.0
go: downloading github.com/fluxio/iohelpers v0.0.0-20160419043813-3a4dd67a94d2
go: downloading github.com/spf13/cast v1.3.0
go: downloading github.com/shurcooL/sanitized_anchor_name v1.0.0
go: downloading github.com/mitchellh/mapstructure v1.1.2
???é?	öSU?8??6%?/,??,A??i?A*)???EQ?2Z?Z7@H?(jk?g?Vw?Q??u?m?K
                                                            *????FY
??T?+???D????Y?̬a???d??_????;?s?@????9g?y핃bEE???I?????O?=<?}U?$??I?+Mx?,??????h?)??	??g?hJ??C9???~3E=??d??ɢ??2??^v???ŵ_
           ??昞;c~ö?R̯??????3w????x???Xic???Q?ü"?w?G???9?Ü??g?좼????ZD/:?z??????xnK?7???/MK??t?Ü??O2h???N??n<류?ָ??b?Ü?Ä??zUI??q??E?z1?x??????????Jö????s$?????s?:ݴ?'??????üur?üR??å??Hd?V߾?׷??????Nq??Å?a?P?T?/??????Q??P?? [... cut ...] m?s?Lo??%ٗÅöٰK?κ?RS?u?S???)?̷ߢ??I5??u???E5üLYs???w̽z??y7
                                                                                    L??????öKÜéB?;0??W?W@????ö?ÜÅ??C?<??L櫷)???????cÅQ?%?r?'ZS=?R??M?'??a????	?aw?rOrf?4Y???/?????C??G?s?t??cl?-Q?5öYéT?*???!?
                                                                        ö=P?¶?͗	?hIyPÜ?ö9??0?Z#?+?a?H?????A?????W3œ???)?Lq?g???t???ryÜz?<ĝÜ(??>Ä?4???C??ÜE?o?/???ߥߵ??4?'Y0?g?1??
?d:???e?N?X????/?,X?ü?_??S?-L??1/݂j	ü?d??t+
...
```

[image]

4ntoine, 2021-07-26:
Any assistance needed from my side on this?

4ntoine, 2021-07-26:
having the same issue with 0.18.2

silvanocerza, 2021-07-26:
> Any assistance needed from my side on this?

Am not working on this right now really, we have other things with higher priority and this is not a huge blocking issue so it doesn't concern me much for now. 
If I'll need any info from your side I'll let you know.

Thanks for reporting it by the way. :pray:

4ntoine, 2021-10-04:
I've just tried to build "0.19.0" on mac Big Sur 11.6 with Docker version 20.10.7, build f0df350 and having the same issue:
```
...
go: downloading github.com/xanzy/ssh-agent v0.2.1
go: downloading gopkg.in/warnings.v0 v0.1.2
‹
...
```
[build_linux_64bit.2.log.zip](https://github.com/arduino/arduino-cli/files/7280382/build_linux_64bit.2.log.zip)

4ntoine, 2021-10-05:
same for `task dist:macOS_64bit`

silvanocerza, 2022-02-10:
This might be related: actions/virtual-environments#2619.

Further investigation is required.

umbynos, 2022-12-14:
Hi @4ntoine does this still happens?