arduino/arduino-cli #1378: arduino-cli: not found → Installation failed in alpine docker container

robbyHuelsi, 2021-08-07:
Hello,
I want to install arduino-cli in a docker container of the [alpine image](https://hub.docker.com/_/alpine). To create, start and interact with such a container I type `docker run -it alpine`.
### Current behavior
Now I try to install arduino_cpi inside the container using the [install script](https://arduino.github.io/arduino-cli/latest/installation/#use-the-install-script):
```
/ # apk add --update --no-cache curl
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/x86_64/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/x86_64/APKINDEX.tar.gz
(1/5) Installing ca-certificates (20191127-r5)
(2/5) Installing brotli-libs (1.0.9-r5)
(3/5) Installing nghttp2-libs (1.43.0-r0)
(4/5) Installing libcurl (7.78.0-r0)
(5/5) Installing curl (7.78.0-r0)
Executing busybox-1.33.1-r3.trigger
Executing ca-certificates-20191127-r5.trigger
OK: 8 MiB in 19 packages
/ # curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=/usr/bin sh
Installing in /usr/bin
ARCH=64bit
OS=Linux
Using curl as download tool
TAG=0.18.3
CLI_DIST=arduino-cli_0.18.3_Linux_64bit.tar.gz
Downloading https://downloads.arduino.cc/arduino-cli/arduino-cli_0.18.3_Linux_64bit.tar.gz
sh: /usr/bin/arduino-cli: not found
Failed to install arduino-cli
``` 
Installation failed but the new command is available:
```
/ # which arduino-cli
/usr/bin/arduino-cli
```
However, the cli cannot be used:
```
/ # arduino-cli version
/bin/sh: arduino-cli: not found
```

### Expected behavior
I found an [repository](https://github.com/zmarcantel/arduino-docker/blob/master/Dockerfile) using an old version of cli (0.3.7-alpha.preview - due to the old download url as described in the deprecation notice [here](https://arduino.github.io/arduino-cli/latest/installation/#download)).  Following this repository I type for installing:
```
/ # apk add --update --no-cache wget
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/x86_64/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/community/x86_64/APKINDEX.tar.gz
(1/3) Installing libunistring (0.9.10-r1)
(2/3) Installing libidn2 (2.3.1-r0)
(3/3) Installing wget (1.21.1-r1)
Executing busybox-1.33.1-r3.trigger
OK: 8 MiB in 17 packages
/ # wget -O /tmp/a-cli.tar.bz2 https://downloads.arduino.cc/arduino-cli/arduino-cli-latest-linux64.tar.bz2
--2021-08-07 21:35:32--  https://downloads.arduino.cc/arduino-cli/arduino-cli-latest-linux64.tar.bz2
Resolving downloads.arduino.cc (downloads.arduino.cc)... 104.18.28.45, 104.18.29.45, 2606:4700::6812:1c2d, ...
Connecting to downloads.arduino.cc (downloads.arduino.cc)|104.18.28.45|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9553825 (9.1M) [application/x-tar]
Saving to: '/tmp/a-cli.tar.bz2'

/tmp/a-cli.tar.bz2                              100%[=======================================================================================================>]   9.11M  30.3MB/s    in 0.3s    

2021-08-07 21:35:33 (30.3 MB/s) - '/tmp/a-cli.tar.bz2' saved [9553825/9553825]

/ # mkdir /tmp/a-cli && tar -xvf /tmp/a-cli.tar.bz2 -C /tmp/a-cli && mv /tmp/a-cli/* /usr/bin/arduino-cli
arduino-cli
```
Et voilà! It works. But...
```
/ # arduino-cli version
arduino-cli Version: 0.3.7-alpha.preview Commit: 69d86a4 BuildDate: 2021-08-07 21:42:34.9162984 +0000 UTC
```
 ... it's actually a very old alpha version. 😟

### Environment

- Intel MacBook Pro 2018, macOS 11.5.1
- Alpine version (`uname -a`): `Linux be401c95d0a6 5.10.25-linuxkit #1 SMP Tue Mar 23 09:27:39 UTC 2021 x86_64 Linux`
- CLI version: 0.18.3

Thanks a lot for your help! 🙏
Best,
Robert

per1234, 2021-08-09:
>it's actually a very old alpha version.

The reason for this is explained here:
https://arduino.github.io/arduino-cli/latest/installation/#latest-packages

>**Deprecation notice**: links in the form http://downloads.arduino.cc/arduino-cli/arduino-cli-latest-<platform>.tar.bz2 won’t be further updated. That URL will provide the version 0.3.7-alpha.preview, regardless of further releases.

The new URL is https://downloads.arduino.cc/arduino-cli/arduino-cli_latest_Linux_64bit.tar.gz

silvanocerza, 2021-08-09:
@robbyHuelsi run this first, it worked for me.

```
apk add libc6-compat
```

From this [StackOverflow answer](https://stackoverflow.com/questions/66963068/docker-alpine-executable-binary-not-found-even-if-in-path).

robbyHuelsi, 2021-09-08:
That's it! Thanks for the quick solution 😊

closed 2021-09-08 by robbyHuelsi