arduino/arduino-cli #1401: `board search` fails with old core and `core list` not show latest version

umbynos, 2021-08-19:
## Bug Report
I think that those two issues below are somehow related, that's why I'm creating a single issue 

### Current behavior
When I install an older version of a core (and also when a new core is released): 
1. the `board search` command returns an empty list
2. the Latest field in the `core list` commands always shows the currently installed version and not the actual latest.

``` bash
$ arduino-cli core list
ID                Installed Latest Name                                             
         
                   
$ arduino-cli core install arduino:samd@1.8.10
Tool arduino:arm-none-eabi-gcc@7-2017q4 already installed
Downloading packages...
arduino:bossac@1.7.0-arduino3 downloaded                                                                                                                                             
arduino:openocd@0.10.0-arduino7 downloaded                                                                                                                                           
arduino:CMSIS@4.5.0 downloaded                                                                                                                                                       
arduino:CMSIS-Atmel@1.2.0 downloaded                                                                                                                                                 
arduino:arduinoOTA@1.2.1 downloaded                                                                                                                                                  
arduino:samd@1.8.10 downloaded                                                                                                                                                       
Installing arduino:bossac@1.7.0-arduino3...
arduino:bossac@1.7.0-arduino3 installed
Installing arduino:openocd@0.10.0-arduino7...
arduino:openocd@0.10.0-arduino7 installed
Installing arduino:CMSIS@4.5.0...
arduino:CMSIS@4.5.0 installed
Installing arduino:CMSIS-Atmel@1.2.0...
arduino:CMSIS-Atmel@1.2.0 installed
Installing arduino:arduinoOTA@1.2.1...
arduino:arduinoOTA@1.2.1 installed
Installing arduino:samd@1.8.10...
Configuring platform...
arduino:samd@1.8.10 installed

$ arduino-cli core list # Latest should show 1.8.11
ID                Installed Latest Name                                        
arduino:samd      1.8.10    1.8.10 Arduino SAMD Boards (32-bits ARM Cortex-M0+)

$ arduino-cli board search arduino:samd:mkrwifi1010 # should list something
Board Name FQBN Platform ID 
                                                              <-- empty

$ arduino-cli core uninstall arduino:samd
Uninstalling arduino:samd@1.8.10...
arduino:samd@1.8.10 uninstalled
Uninstalling arduino:bossac@1.7.0-arduino3, tool is no more required...
arduino:bossac@1.7.0-arduino3 uninstalled
Uninstalling arduino:openocd@0.10.0-arduino7, tool is no more required...
arduino:openocd@0.10.0-arduino7 uninstalled
Uninstalling arduino:CMSIS@4.5.0, tool is no more required...
arduino:CMSIS@4.5.0 uninstalled
Uninstalling arduino:CMSIS-Atmel@1.2.0, tool is no more required...
arduino:CMSIS-Atmel@1.2.0 uninstalled
Uninstalling arduino:arduinoOTA@1.2.1, tool is no more required...
arduino:arduinoOTA@1.2.1 uninstalled

$ arduino-cli core install arduino:samd
Tool arduino:arm-none-eabi-gcc@7-2017q4 already installed
Downloading packages...
arduino:bossac@1.7.0-arduino3 already downloaded
arduino:openocd@0.10.0-arduino7 already downloaded
arduino:CMSIS@4.5.0 already downloaded
arduino:CMSIS-Atmel@1.2.0 already downloaded
arduino:arduinoOTA@1.2.1 already downloaded
arduino:samd@1.8.11 downloaded                                                                                                                                                       
Installing arduino:bossac@1.7.0-arduino3...
arduino:bossac@1.7.0-arduino3 installed
Installing arduino:openocd@0.10.0-arduino7...
arduino:openocd@0.10.0-arduino7 installed
Installing arduino:CMSIS@4.5.0...
arduino:CMSIS@4.5.0 installed
Installing arduino:CMSIS-Atmel@1.2.0...
arduino:CMSIS-Atmel@1.2.0 installed
Installing arduino:arduinoOTA@1.2.1...
arduino:arduinoOTA@1.2.1 installed
Installing arduino:samd@1.8.11...
Configuring platform...
arduino:samd@1.8.11 installed

$ arduino-cli core list
ID                Installed Latest Name                                        
arduino:samd      1.8.11    1.8.11 Arduino SAMD Boards (32-bits ARM Cortex-M0+)

$ arduino-cli board search arduino:samd:mkrwifi1010
Board Name            FQBN                     Platform ID  
Arduino MKR WiFi 1010 arduino:samd:mkrwifi1010 arduino:samd

```

### Expected behavior
1. `board search` is expected to work with every core version
2. `board list` Latest field should show the latest core version available 

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-26T07:39:22Z` but the problem seems to be present also from `0.17.0` (I did not try earlier versions)
- OS and platform: Both Ubuntu 18.04 and MacOS Big Sur

### Additional context
I tried to debug the problem, It seems it's not a visualization problem, but a init related one

per1234, 2021-08-19:
I think this also affects Arduino IDE 2.x because I found that there are no boards if you install any version of a platform other than the latest.

closed 2021-08-25 by silvanocerza

silvanocerza, 2021-08-25:
Fixed by #1409 and #1410.