arduino/arduino-cli #1408: arduino-cli: uploads hex file successfully, generates an error message.

mtulu-argeX, 2021-08-24:
We use arduino-cli to upload hex files to Arduino Leonardo based hardware through our programming software based on Ardu Blockly. 

When we hit upload, arduino-cli successfully uploads the hex file however it generates an error. Here is the error output:

`
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/jack/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/Users/jack/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem14301
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file "/Users/jack/Desktop/watchX_Blocks/hex-data/Pacman.hex"
avrdude: writing flash (27374 bytes):

Writing | ################################################## | 100% 2.07s

avrdude: 27374 bytes of flash written

avrdude done.  Thank you.

`

We suspect that arduino-cli tries to upload hex file a couple of times. It uploads the hex in 2nd or 3rd try, because it fails on first one it generates an error.

per1234, 2021-08-24:
Hi @mtulu-argeX. Thanks for your interest in this open source project. The output you have shared is the output from a successful AVRDUDE upload. This is not an error. So you can carry on with your  use of Arduino CLI without concern.

Regards, Per

closed 2021-08-24 by per1234

mtulu-argeX, 2021-08-24:
Hi @per1234 ,

Thank you for your fast response. It indeed uploads the hex successfully. We receive the stream output as stderr. Any idea why Arduino-cli gives the output as stderr?

All my bests, Mustafa.

per1234, 2021-08-24:
Because this is the stream used by the AVRDUDE uploader tool. You'll get the same result if you run the `avrdude` command directly.

As for why AVRDUDE does this, I believe it is explained here:
https://www.nongnu.org/avrdude/environment.html

mtulu-argeX, 2021-08-24:
The link was very useful. Thank you so much!