arduino/arduino-cli #1419: panic: runtime error: invalid memory address or nil pointer dereference

Goral64, 2021-08-30:
## Bug Report

### Current behavior
I am trying to create a docker image with arduino-cli but with the command:

RUN arduino-cli lib update-index

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0xa8 pc=0x7318bf]

goroutine 1 [running]:
github.com/leonelquinteros/gotext.(*Domain).Get(0x0, {0xe4f074, 0x19}, {0x0, 0x0, 0x0})
        /go/pkg/mod/github.com/leonelquinteros/gotext@v1.5.0/domain.go:253 +0x5f
github.com/leonelquinteros/gotext.(*Po).Get(...)
        /go/pkg/mod/github.com/leonelquinteros/gotext@v1.5.0/po.go:84
github.com/arduino/arduino-cli/i18n.Tr({0xe4f074, 0xc00030b810}, {0x0, 0xc000340ed0, 0x15b9948})
        /go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210830142620-ff4eb92a66a9/i18n/i18n.go:50 +0x45
github.com/arduino/arduino-cli/arduino/libraries/librariesmanager.init()
        /go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210830142620-ff4eb92a66a9/arduino/libraries/librariesmanager/install.go:38 +0xc6

### Expected behavior

INFO[0000] Using config file: /root/.arduino15/arduino-cli.yaml
INFO[0000] arduino-cli version 0.0.0-git
[...]
### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.0.0-git Commit:  Date:
- OS and platform: Ubuntu 20.04.2 LTS (Focal Fossa)

### Additional context
Image builds fine on another server Ubuntu 20.04.1 LTS (Focal Fossa)

per1234, 2021-08-31:
I can see the problem here:
```
/go/pkg/mod/github.com/leonelquinteros/gotext@v1.5.0
```
Arduino CLI uses `github.com/leonelquinteros/gotext@v1.4.0`. The dependency [is pinned to that version](https://github.com/arduino/arduino-cli/blob/ff4eb92a66a959e45ed36b4e43a183c69ab3360a/go.sum#L157). Did you do something like this?:
```
go get github.com/leonelquinteros/gotext@latest
```

Minimal demonstration of the issue:
```go
package main

import (
	"github.com/arduino/arduino-cli/i18n"
)

func main() {
	i18n.Tr("foo")
}
```
It seems that we will need to make this change when the time comes to update to the new version:
https://pkg.go.dev/github.com/leonelquinteros/gotext#NewPo
>NewPo should always be used to instantiate a new Po object

```diff
--- a/i18n/locale.go
+++ b/i18n/locale.go
@@ -32,7 +32,7 @@ var (
 )

 func init() {
-       po = new(gotext.Po)
+       po = gotext.NewPo()
 }

 func initRiceBox() {
@@ -79,6 +79,6 @@ func findMatchingLocale(locale string, supportedLocales []string) string {

 func setLocale(locale string) {
        poFile := box.MustBytes(locale + ".po")
-       po = new(gotext.Po)
+       po = gotext.NewPo()
        po.Parse(poFile)
 }
```

silvanocerza, 2021-08-31:
Did you make some custom modifications and built the Arduino CLI yourself? The `version` output is not the one I'd expect from one of our builds.

Goral64, 2021-08-31:
My Dockerfile:

```
# Stage 1 of 2
FROM golang:latest
RUN go get -u github.com/arduino/arduino-cli

# Stage 2 of 2
FROM ubuntu:20.04

ENV DEBIAN_FRONTEND=noninteractive

COPY --from=0 /go/bin/arduino-cli /usr/bin/arduino-cli
ADD cli-config.yml /root/.arduino15/arduino-cli.yaml

# arduino
RUN apt-get update \
    && apt-get install -yy --no-install-recommends arduino \
    && apt-get -y autoremove \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*

RUN arduino-cli lib update-index \
		&&arduino-cli lib install \
		"Adafruit BME280 Library" \
		"Adafruit Si7021 Library" \
		"Adafruit Unified Sensor" \
		"ClosedCube SHT31D" \
		"DallasTemperature" \
		"DHT sensor library" \
		"DoubleResetDetector" \
		"HLW8012" \
		"OneWire" \ 
		&& arduino-cli core update-index \ 
		&& arduino-cli core install esp8266:esp8266 \
		&& arduino-cli core install esp32:esp32

COPY script/run.sh /run.sh
RUN chmod 755 /run.sh

CMD ["/run.sh"]
```
My cli-config.yml file
```
board_manager:
  additional_urls:
    - http://arduino.esp8266.com/stable/package_esp8266com_index.json
    - https://dl.espressif.com/dl/package_esp32_index.json
```
I didn't change anything in the arduino-cli code

silvanocerza, 2021-08-31:
```
RUN go get -u github.com/arduino/arduino-cli
```
This line might be the issue, `go help get` says this:

> The -u flag instructs get to update modules providing dependencies
of packages named on the command line to use newer minor or patch
releases when available.

Try changing it to:

```
RUN go install github.com/arduino/arduino-cli
```

Goral64, 2021-08-31:
`RUN go install github.com/arduino/arduino-cli@latest`

Thank you. This change helped. The image is built without errors.

Goral64, 2021-08-31:
Resolved

closed 2021-08-31 by Goral64