arduino/arduino-cli #1421: Compiling with --build-path

josejsarmento, 2021-08-31:
## Bug Report

### Current behavior

Compiling a sketch with the `--build-path` flag is not working for me. Whenever I run the following command:

`arduino-cli compile --build-path ../ArduinoOutput -b esp8266:esp8266:generic sketch`

I get:

```
Couldn't deeply cache core build: Rel: can't make ../ArduinoOutput relative to /var/folders/gg/v110hgf95ls12d5vvmv2mc0c0000gn/T/arduino-core-cache
Running normal build of the core...
```

### Expected behavior

I expect the program to save the compiled files at the chosen folder and to re-use them when I compile again afterwards. This would be similar to the 'output' option in the Arduino extension in Visual Studio Code.

### Environment

- CLI version (output of `arduino-cli version`): `arduino-cli alpha Version: 0.18.3 Commit: d710b642 Date: 2021-05-21T17:11:27Z`
- OS and platform: Mac OS 11.5.2

josejsarmento, 2021-08-31:
I've found the issue: arduino-cli doesn't handle relative paths
https://github.com/arduino/arduino-cli/issues/630#issue-585891619

silvanocerza, 2021-08-31:
Duplicate of #630, closing.

closed 2021-08-31 by silvanocerza