arduino/arduino-cli #1437: Command `config add` for additional URLs does not check presence of URL in `board_manager` tree

ubidefeo, 2021-09-08:
**Describe the bug**
When manually adding an additional url via `arduino-cli config add ...` the command does not check for existing matching entries in the list, hence generating multiple entries for the same URL

**To Reproduce**
Steps to reproduce the behavior:
1. `arduino-cli config add board_manager.additional_urls https://my.url/package_mycustom_index.json`
2. `arduino-cli config dump`
3. verify that our URL is at the end of the `board_manager` group of `additional_urls`
4. `arduino-cli config add board_manager.additional_urls https://my.url/package_mycustom_index.json`
5. `arduino-cli config dump`
6. verify that the URL is twice at the end of the `board_manager` group of `additional_urls`
7. `arduino-cli config remove board_manager.additional_urls https://my.url/package_mycustom_index.json`
8. `remove` will remove any istance of said URL

**Expected behavior**
The command should check for the presence of said URL and not add it if already there

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Version: 11.5.2

**Additional context**
n/a

ubidefeo, 2021-09-10:
@per1234 @silvanocerza 
of course it was :D

henrygab, 2022-10-02:
Almost a year.  I also hit this.

**Question: If a PR was presented to fix this, what is the preferred fix?**

Would it be preferred to only apply to board manager URLs,
or to all config settings that might take multiple values?

PR #1902 (2022-10-05): Prevent duplicate entries in board_manager.additional_urls
Fixes #1437.

**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] ~Docs have been added / updated (for bug fixes / features)~
- [ ] ~`UPGRADING.md` has been updated with a migration gu [... cut]
closed 2022-10-05 by cmaglie