arduino/arduino-cli #1439: Different import zip file behaviour (gRPC vs IDE).

4ntoine, 2021-09-09:
## Bug Report

### Current behavior

Download repo as a zip from https://github.com/phyphox/phyphox-arduino and use downloaded file ("phyphox-arduino-master.zip") as `zipLibraryInstall` path.

Calling `zipLibraryInstall()` fails with "Failed to install the li: gRPC Error (code: 13, codeName: INTERNAL, message: Library install failed: library is not valid: missing header file "phyphox-arduino-master.h", details: [], rawResponse: null, trailers: {})".

I've checked i can import exactly the same file in IDE.

### Expected behavior

The lib is imported.

### Environment

- CLI version (output of `arduino-cli version`): 0.19.0
- OS and platform: macOS 10.14.6

### Additional context

per1234, 2021-09-09:
Hi @4ntoine. Thanks for taking the time to submit this issue.

I see we already have a prior issue report about this at https://github.com/arduino/arduino-cli/issues/1306.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-09-09 by per1234

4ntoine, 2021-09-09:
Agreed, thank you.