arduino/arduino-cli #1444: stm32: Error during Upload: Property 'upload.tool.serial' is undefined

cmaglie, 2021-09-10:
I noticed a regression that I expect is caused by this PR. When using arduino-cli from git now, uploads using (our slightly customized version of) the [STM32 core](https://github.com/stm32duino/Arduino_Core_STM32) fail with:

    Error during Upload: Property 'upload.tool.serial' is undefined

The board in question indeed does not define `upload.tool.serial`, but still uses the pre-pluggable syntax of defining `upload.tool`, which should be translated to `upload.tool.default` by `convertUploadToolsToPluggableDiscovery()`. However, this core does not define `upload.tool` on the board directly, but defines it in a board menu: https://github.com/stm32duino/Arduino_Core_STM32/blob/974b35690cbf4680ce5746c61c69fb20935bae2f/boards.txt#L522

To reproduce:

```
$ arduino-cli compile --fqbn STMicroelectronics:stm32:Nucleo_32:pnum=NUCLEO_F031K6,upload_method=serialMethod --upload --port /dev/ttyACM0 
Sketch uses 7612 bytes (23%) of program storage space. Maximum is 32768 bytes.
Global variables use 828 bytes (20%) of dynamic memory, leaving 3268 bytes for local variables. Maximum is 4096 bytes.
Error during Upload: Property 'upload.tool.serial' is undefined
```

I guess this is fairly easy to fix in the core, though (just define both `upload.tool.default` (or `upload.tool.<protocol>`) in addition to the legacy `upload.tool`, but it might be useful to fix compatibility here too. An obvious fix would be to implement the fallback when actually doing the upload, rather than when loading the boards).

@fpistm, you might find this interesting too :-)

_Originally posted by @matthijskooijman in https://github.com/arduino/arduino-cli/pull/1333#issuecomment-916901069_

fpistm, 2021-09-10:
Thanks @matthijskooijman and @cmaglie 
Let me know if I need to update the core.

maxdignan, 2021-10-13:
@cmaglie @fpistm Can either of you help me figure out how to edit this? I'm struggling at finding the right files to edit. Thanks!

per1234, 2021-10-13:
@maxdignan we should restrict the discussion here to the Arduino CLI code base. Please post support questions to the Arduino forum:
https://forum.arduino.cc/

I'm sure we'll be able to help you out over there.

per1234, 2021-12-18:
The same bug affects [the `bootloader.tool` property](https://arduino.github.io/arduino-cli/dev/platform-specification/#burn-bootloader) as well.

Real life example:
https://github.com/SpenceKonde/ATTinyCore/blob/1.5.2/avr/boards.txt#L603
```
$ arduino-cli burn-bootloader -b ATTinyCore:avr:attinyx5micr:burnmode=install -P usbasp
Error during Upload: Property 'bootloader.tool.default' is undefined
```

cmaglie, 2022-01-24:
The fix will be released with arduino-cli >=0.21.0 (or you can try the nightly as soon as it updates)

closed 2022-01-24 by cmaglie

cjgu00, 2022-07-09:
I had the similar error for SeeedStudio XIAO BLE. Are there any helps?

Using board '**xiaonRF52840**' from platform in folder: C:\Users\R11117_NHRI\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\2.6.1
Using core 'arduino' from platform in folder: C:\Users\R11117_NHRI\AppData\Local\Arduino15\packages\Seeeduino\hardware\nrf52\2.6.1
Detecting libraries used...
"C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/defines.txt" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_SEEED_XIAO_NRF52840 -DARDUINO_ARCH_NRF52 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/includes.txt" "C:\\Users\\R11117_NHRI\\AppData\\Local\\Temp\\arduino-sketch-C50A6AEBE81FFD7200ACC4A3D2D25ACB\\sketch\\FadeLED_XiaoBle.ino.cpp" -o nul
Alternatives for RGBLed.h: [RGB@1.0.9]
ResolveLibrary(RGBLed.h)
  -> candidates: [RGB@1.0.9]
"C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/defines.txt" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_SEEED_XIAO_NRF52840 -DARDUINO_ARCH_NRF52 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840" "-Id:\\User\\Arduino\\libraries\\RGB\\src" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/includes.txt" "C:\\Users\\R11117_NHRI\\AppData\\Local\\Temp\\arduino-sketch-C50A6AEBE81FFD7200ACC4A3D2D25ACB\\sketch\\FadeLED_XiaoBle.ino.cpp" -o nul
Using cached library dependencies for file: d:\User\Arduino\libraries\RGB\src\RGBLed.cpp
Generating function prototypes...
"C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/defines.txt" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_SEEED_XIAO_NRF52840 -DARDUINO_ARCH_NRF52 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840" "-Id:\\User\\Arduino\\libraries\\RGB\\src" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated" "-IC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino/api/deprecated-avr-comp" "-iprefixC:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\cores\\arduino" "@C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\hardware\\nrf52\\2.6.1\\variants\\SEEED_XIAO_NRF52840/includes.txt" "C:\\Users\\R11117_NHRI\\AppData\\Local\\Temp\\arduino-sketch-C50A6AEBE81FFD7200ACC4A3D2D25ACB\\sketch\\FadeLED_XiaoBle.ino.cpp" -o "C:\\Users\\R11117_NHRI\\AppData\\Local\\Temp\\arduino-sketch-C50A6AEBE81FFD7200ACC4A3D2D25ACB\\preproc\\ctags_target_for_gcc_minus_e.cpp" [... cut ...]
"C:\\Users\\R11117_NHRI\\AppData\\Local\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-size" -A "C:\\Users\\R11117_NHRI\\AppData\\Local\\Temp\\arduino-sketch-C50A6AEBE81FFD7200ACC4A3D2D25ACB/FadeLED_XiaoBle.ino.elf"
Sketch uses 86040 bytes (10%) of program storage space. Maximum is 811008 bytes.
Global variables use 44024 bytes (18%) of dynamic memory, leaving 193544 bytes for local variables. Maximum is 237568 bytes.
**Property 'upload.tool.serial' is undefined**

per1234, 2022-07-09:
@cjgu00 please post over on the Arduino forum. I'll help you out over there:

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

cjgu00, 2022-07-12:
I fixed my bug by following the steps described by ptillisch,

https://forum.arduino.cc/t/arduino-2-beta-12-stlink-programmer-error/921351/6

It do work! Thanks very much to give me the hint to find information on the forum.

commit 569e1944 (2022-01-24): Improved pluggable discovery conversion for platforms not supporting it (#1629)

* Factored property composition in convertUploadToolsToPluggableDiscovery

* Auto-generated properties are cached and added after cycling on the original map

* Autogenerated upload tool properties are searched in sub-configs too

Fix #1444

* Added STMicroelectronics:stm32:Nucleo_32:pnum=NUCLEO_F031K6 to upload mock tests

* Use composed board properties to detect user fields

Some platforms may add information through the optional config part of
the FQBN (platforms menu items).