arduino/arduino-cli #1445: Crash on build from source because ioctl panic bug (go 1.17)

judaew, 2021-09-10:
## Bug Report

### Current behavior

Crash during build from source in MacPorts in `0.19.0` version of arduino-cli. This is being caused by an ioctl panic bug in golang.org/x/sys: golang/go#47776.

Logs build:
- [error-arduino-cli.txt](https://github.com/arduino/arduino-cli/files/7145539/panic-arduino-cli.txt)
- https://dev.azure.com/macports/macports-ports/_build/results?buildId=16632&view=logs&j=ca395085-040a-526b-2ce8-bdc85f692774

See more:
- https://github.com/macports/macports-ports/pull/12185

### Expected behavior

Fixes that will fix the arduino-cli build with go 1.17.

### Environment

- ~~CLI version (output of `arduino-cli version`):~~
- OS and platform: macOS 11.5.2 20G95 x86_64

per1234, 2021-09-11:
Hi @judaew. I believe this issue was fixed by https://github.com/arduino/arduino-cli/pull/1427, which was merged after the 0.19.0 release of Arduino CLI. Please try building again using the code from the tip of the `master` branch and then let us know if the issue is resolved.

per1234, 2021-09-11:
By the way, we are using Go 1.16 for all development and testing of Arduino CLI. The use of 1.17 for beta testing is welcome, as it can help us to prepare for an eventual transition to using that version, but please don't use 1.17 for any sort of production deployment of Arduino CLI.

judaew, 2021-09-11:
Thank you for the quick response. I confirm, this PR fixes the port build with go 1.17.

Unfortunately, I can't use go 1.16 to build arduino-cli as recommended because MacPorts like Arch Linux uses the latest version of packages to build.

closed 2021-09-11 by judaew