arduino/arduino-cli #1450: Cannot load Esp32

vedantroy, 2021-09-13:
## Bug Report

### Current behavior

When running any arduino-cli command, I get the following error:
```
Error initializing instance: loading platform release esp32:esp32@2.0.0: loading boards: skipping loading of boards esp32:esp32:deneyapmini, esp32:esp32:atmegazero_esp32s2: malformed custom board options
Error initializing instance: loading platform release espressif:esp32@2.0.0: loading boards: skipping loading of boards espressif:esp32:deneyapmini, espressif:esp32:atmegazero_esp32s2: malformed custom board options
```

### Expected behavior

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.19.0 Commit: 56419ecd Date: 2021-09-02T14:47:35Z

- OS and platform:

(Output of `lsb_release -a`)

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

### Additional context
I think it might be related to this:

https://github.com/espressif/arduino-esp32/pull/5623

per1234, 2021-09-13:
Yes, this is related to https://github.com/arduino/arduino-cli/pull/1326, but actually the configuration like in https://github.com/espressif/arduino-esp32/issues/5618 caused problems even before that, only before the effects were isolated to specific boards rather than the entire platform, and with a much more cryptic symptom than the current clear error message.

silvanocerza, 2021-10-04:
espressif/arduino-esp32#5673 superseded espressif/arduino-esp32/pull/5623 and has been already merged.

This is not an Arduino CLI issue though, we avoid loading boards that do not respect the platform specifications. That's a task for that platform developer to solve.

Closing.

closed 2021-10-04 by silvanocerza

brentru, 2021-10-04:
@silvanocerza  Re-opening this issue. Even with building on `arduino-esp32` master (which includes https://github.com/espressif/arduino-esp32/pull/5673), I am still getting the following output from Arduino-CLI:

```
arduino-cli compile --fqbn esp32:esp32:adafruit_funhouse_esp32s2:CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,PSRAM=enabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none --warnings all --export-binaries examples/sketch.ino
Error initializing instance: loading platform release esp32:esp32@2.0.0: loading boards: skipping loading of boards esp32:esp32:deneyapmini, esp32:esp32:atmegazero_esp32s2: malformed custom board options
Error initializing instance: loading platform release espressif:esp32@2.0.0: loading boards: skipping loading of boards espressif:esp32:deneyapmini: malformed custom board options
```

Any ideas?