arduino/arduino-cli #1451: Q: how to pass additional boards managers URLs?

4ntoine, 2021-09-14:
## Bug Report

### Current behavior

There is no "additional_urls" argument in [`BoardsListAllRequest`](https://github.com/arduino/arduino-cli/blob/826adc5d80bd48938908850cc9d10f8785b9fdd3/rpc/cc/arduino/cli/commands/v1/board.proto#L191).

### Expected behavior

I'd expect `repeated string additional_urls` argument in `BoardsListAllRequest` to be able to add additional boards (eg. esp32, esp8266). I can do it IDE so i guess it should be possible via gRPC.

### Environment

- CLI version (output of `arduino-cli version`): "0.19.0"
- OS and platform:

### Additional context

Any other way one can do it?

4ntoine, 2021-09-14:
Arduino CLI has `--additional-urls strings   Additional URLs for Boards Manager.` cli argument so i guess it should be exposed via gRPC too

4ntoine, 2021-09-14:
Ahh, it seems it's meant to be set in the [settings](https://arduino.github.io/arduino-cli/0.19/configuration). How is it expected to work in the following case: there is a list of url (or just empty). In order to find a core for the board one needs to modify a list of URLs in the setting, do the search and then restore to it's original state?

silvanocerza, 2021-09-14:
Yep, I wouldn't restore the original state though, you could leave it like that. You'll probably also need to update the indexes with `UpdateIndex` and then `Init` again.

4ntoine, 2021-09-14:
@silvanocerza thank you for the explanation

closed 2021-09-14 by 4ntoine