arduino/arduino-cli #1453: Refactor cli package to use a global instance

silvanocerza, 2021-09-15:
## Enhancement

An instance keeps track of all the information needed by the CLI to correctly run all its commands and gRPC functions. The `commands` package keeps track of all the instances created, the CLI needs only one instance to work correctly apart when running as `daemon` mode. When running in `daemon` mode the clients connected to it can create any number of instances.

As of now all commands of the `cli` package handle the creation and the initialization of an instance whenever it's necessary. This behaviour is fine since all commands, apart from `daemon`, are "one shot", they're executed and exit when done. 

Until now there have been no cases in which we would have benefited from reusing an already created and initialized instance when running the CLI in non-daemon mode, but in future we might need it to efficiently support interactivity and similar things. 

For example to implement [#1452](https://github.com/arduino/arduino-cli/issues/1452) in an efficient manner we'll need to make this refactoring happen, otherwise the feature will be completely useless and even annoying for users with lots of platforms and/or libraries. Since as of now we create and initialize a new instance every time the user asks for `--fqbn` completion in `compile`.

silvanocerza, 2022-01-04:
After some investigation I discovered that using a global instance won't speed up dynamic auto completion. 
It should have been obvious really but I didn't think that auto completion works by running an hidden `__complete`, obviously that creates a new process of the `arduino-cli` so a new instance, etc. 
The best solution to speed completion would be to cache that information somewhere in the filesytem and write/read it if necessary.

Nonetheless having a single instance in the Arduino CLI will be beneficial for other reasons.

closed 2022-04-06 by ubidefeo

per1234, 2022-04-06:
@ubidefeo, for future reference, please add a summary of the reason for closure.