arduino/arduino-cli #1459: Wron header name

FStefanni, 2021-09-20:
## Bug Report

### Current behavior
```
arduino-cli lib install --git-url https://github.com/FStefanni/arduino-sht.git
```
The command fails with the following log:

```
Error initializing instance: loading platform release esp32:esp32@2.0.0: loading boards: skipping loading of boards esp32:esp32:atmegazero_esp32s2, esp32:esp32:deneyapmini: malformed custom board options
--git-url and --zip-path flags allow installing untrusted files, use it at your own risk.
Enumerating objects: 68, done.
Counting objects: 100% (68/68), done.
Compressing objects: 100% (44/44), done.
Total 68 (delta 32), reused 49 (delta 19), pack-reused 0
Error installing Git Library: Library install failed: library is not valid: missing header file "arduino-sh.h"
```
### Expected behavior

The searched header file should be `arduino-sht.h` and not `arduino-sh.h`.

### Environment

- CLI version (output of `arduino-cli version`): arduino-cli alpha Version: 0.19.0 Commit: 56419ecd Date: 2021-09-02T14:47:35Z
- OS and platform: Debian testing (bookworm)

### Additional context

* The library is a clone of [arduino-sht](https://github.com/Sensirion/arduino-sht), on which I am working to create a pr.
* Please notice that I have added the `arduino-sht.h` file.
* If I add the wrong file `arduino-sh.h`, the library is successfully installed.

The content of `library.properties` is:
```
name=arduino-sht
version=1.2.0
author=Johannes Winkelmann, Andreas Brauchli
maintainer=Johannes Winkelmann <jwi@sensirion.com>
sentence=Support for Sensirion's humidity and temperature sensors.
paragraph=Supported sensors: SHTC1, SHTC3, SHTW1, SHTW2, SHT3x-DIS (I2C), SHT3x-ARP, SHT4x
category=Sensors
url=https://developer.sensirion.com
architectures=*
```

per1234, 2021-09-20:
Hi @FStefanni. Thanks for reporting this.

It's related to https://github.com/arduino/arduino-cli/issues/1306 in that this check shouldn't be done at all. However, the "sht" vs. "sh" issue is an additional bug that might indicate there is a separate issue that will persist even after https://github.com/arduino/arduino-cli/issues/1306 is fixed.

FStefanni, 2021-09-20:
Hi,

yes this is surely related, even if not exactly the same.
I hope this will be fixed asap, since it is very blocking.

Regards.

per1234, 2021-09-20:
I don't see any reason it should be blocking. `arduino-cli lib install --git-url` is nothing more than a trivial convenience command. Just use `git clone` to install the library instead. If you don't have Git installed on your computer then download the zip file of the repository from GitHub and use `arduino-cli lib install --zip-path` to install it.

FStefanni, 2021-09-20:
Hi,

it is blocking ***for my workflow*** (of course not in general), because I have a set of scripts using `arduino-cli lib install --git-url` to automatically install/create the environment to run a set of compiling tasks.

With this bug, I have to change all the scripts, or to install manually the lib on the hosts.

Regards.

silvanocerza, 2021-09-23:
Fixed by #1462.

closed 2021-09-23 by silvanocerza