arduino/arduino-cli #1467: Possible VM-specific issue?

4ntoine, 2021-09-23:
## Bug Report

### Current behavior

Connection is closed by the server side

Running cli daemon in VM (tried Alpine linux and Ubuntu as Guest OS in UTM and VirtualBox) with port forwarding 50051->50051.
Requested cli version, connection is closed.

### Expected behavior

Connection is not closed and a version response is received.

### Environment

- CLI version (output of `arduino-cli version`): 0.19.0
- OS and platform: macOS 11.5.2 Big Sur + Ubuntu 18.04

### Additional context

Tried to open Guest OS port 50051 and listen for the input:
```
asvm@asvm:~/bin$ nc -l -p 50051
```
, i can see incoming request (from the test):

```
PRI * HTTP/2.0

SM

�:methodPOST:schemehttp:path6/cc.arduino.cli.commands.v1.ArduinoCoreService/Version
:authoritylocalhost:50051
                         content-typeapplication/grpcttrailers
user-agentdart-grpc/2.0.0
```

So the incoming data (request) is received (not sure whether it's corrupted).
Here is the (test) code (Dart):
```dart
ClientChannel _getChannel() => ClientChannel(
    'localhost',
    port: 50051,
    options: const ChannelOptions(credentials: ChannelCredentials.insecure()),
  );

test('version request',  () async {
    final channel = _getChannel();
    final client = ArduinoCoreServiceClient(channel);
    var versionResponse = await client.version(VersionRequest());
    final version = versionResponse.version;
    print('Arduino cli version is: $version');
    await channel.shutdown();
  });
```

If i run cli on the same host the test passes (so i believe it's ok):

> Arduino cli version is: 0.19.0

So it seems that either request is corrupted (by VM which is less likely) or there is some cli bug?
`--verbose` does not help - like no input is received:
```
asvm@asvm:~/bin$ ./arduino-cli daemon --verbose
INFO[0000] Config file not found, using default values  
INFO[0000] arduino-cli version 0.19.0                   
INFO[0000] Setting up Prometheus metrics on :9090/metrics 
INFO[0000] Starting daemon on TCP address 127.0.0.1:50051 
INFO[0000] Daemon is now listening on 127.0.0.1:50051... 
```

Any other way to turn on (more detailed) debugging output?

4ntoine, 2021-09-23:
Added port forwarding 9090->9090 and it seems to be ok:
```
> curl localhost:9090/metrics
# TYPE daemon_daemon counter
daemon_daemon{installationID="bad262c5-9db2-455e-9432-02679ba82c02",success="true"} 1 1632440241976
```

silvanocerza, 2021-09-24:
Am not sure this is an `arduino-cli` bug, in any case you can add get some more logs with `--log-level trace`.

4ntoine, 2021-09-24:
yeah, i was able to communicate from the test in the VM itself. So VM just does not send the response bytes from from Guest OS port to Host OS port. Thanks for the log level info

4ntoine, 2021-09-24:
I've tried "nc" and tested everything and it seems arduino-cli just does not response for some reason if the connection is not from the host. Also i can't see any debug output, eg. "get version request received". 

Attaching version request bytes:
[version_request.txt](https://github.com/arduino/arduino-cli/files/7228023/version_request.txt)

4ntoine, 2021-10-04:
Shouldn't [`127.0.0.1:50051`](https://github.com/arduino/arduino-cli/blob/fb8931c3e14da79932ed5965244010202efa8e01/cli/daemon/daemon.go#L113) be just `:50051` or `0.0.0.0:50051`?

4ntoine, 2021-10-05:
works as designed

closed 2021-10-05 by 4ntoine