arduino/arduino-cli #1468: arduino-cli doesn't like my package index, but the old IDE accepts it

obra, 2021-09-24:
## Bug Report

### Current behavior

The current arduino-cli silently ignores some of my package index files. The package indexes in question are "git head" index files that let the user install the current main branch of my core without needing to publish an updated package file for every commit.

https://github.com/keyboardio/ArduinoCore-GD32-Keyboardio/blob/main/package_gd32_index.json is the URL to an example board index.

This works just great in the 'classic' IDE's boards manager.

The only error I can get out of --verbose mode is 
`INFO[0000] Checking signature                            error="opening signature file: open /Users/jesse/Library/Arduino15/package_gd32_index.json.sig: no such file or directory" index=/Users/jesse/Library/Arduino15/package_gd32_index.json signatureFile=/Users/jesse/Library/Arduino15/package_gd32_index.json.sig`

However, that doesn't appear to be what causes my core's package to get discarded.

arduino-cli version:
`arduino-cli alpha Version: 0.19.0 Commit: 56419ecd Date: 2021-09-02T14:47:35Z`

Is this a regression from the classic IDE's boards manager or is there some sort of a spec change? If it's the latter, I'd hope that running in --verbose mode, arduino-cli would complain about what causes it to discard the unacceptable platform.

obra, 2021-09-24:
Prodding a little bit, it appears that the "size" key for a platform has become a required key. If 'size' is not included, the platform is silently dropped.  That feels like a bug.

ubidefeo, 2021-10-05:
hi @obra 
Sorry for the late reply, I've been digging into this.

What is annoying is that the Java IDE is way too permissive (as it's always been), but the size and/or checksum are fields you should fill in.
Have you given a go at Arduino Lint?
If you build from master you have access to the new Package Index check as well as documentation for all the rules.
We have to finalise a few things before this gets a release tag.

What I get out of it is this
```
Linting package-index in package_gd32_index.json
ERROR: packages[*].platforms[*].help.online property does not have a valid URL format in platform(s):
         keyboardio:gd32@0.0.1
       See: https://arduino.github.io/arduino-cli/latest/package_index_json-specification/#platforms-definitions
       (Rule IL022)
ERROR: Missing packages[*].platforms[*].checksum property in platform(s):
         keyboardio:gd32@0.0.1
       See: https://arduino.github.io/arduino-cli/latest/package_index_json-specification/#platforms-definitions
       (Rule IL032)
ERROR: Missing packages[*].platforms[*].size property in platform(s):
         keyboardio:gd32@0.0.1
       See: https://arduino.github.io/arduino-cli/latest/package_index_json-specification/#platforms-definitions
       (Rule IL036)

Linter results for project: 3 ERRORS, 0 WARNINGS

-------------------
```

please try to just add the size property and see if it goes through, otherwise add the checksum.
We're still very much in the process of documenting the platform, making sure specs are followed by everyone (starting from within) and randomly opening PRs to repos we bump into which are not compliant at any level.

Let me know if you can get this to work, but we do have a couple of open tasks related to these properties of the Package Index that might get a facelift

u.

obra, 2021-10-05:
Hi @ubidefeo - No worries!

The specific workflow I'm trying to unbreak here is to be able to define a platform/core that automatically installs the current head of the git main/master branch of the repo. GitHub provides an easy way to get a persistent URL to a zipfile of the current state of a repo. Being able to drop a platform json file into the repo pointing at that zipfile has meant that it's been easy to support folks who want to be running the current/dev version of a core but aren't themselves developers of that core. Because that zipfile is autogenerated at request time, there's no easy way to obtain a size or checksum.

With the existing IDE, this just works. With the new, stricter parser in arduino-cli, it does not.

The alternative is to manually create a zipfile on every commit and publish a new version of the package.json file with a checksum and size. If that's the only supported solution going forward, it can be made to work, but it feels like a waste of computing resources :/

I hugely appreciate the work you all are doing to modernize and document the platform and ecosystem. It is absolutely making Arduino better. 

In other projects, the way I've typically dealt with new strictures in config files is with versioning, but there may be few enough arduino platforms out there that that'd be overkill here. It *would* be really nice to have a way to say "hey, this core file is dynamically generated, so you're not getting a checksum or size", though.

ubidefeo, 2021-10-05:
I completely understand your motivations and your point.
I'm not 100% sure how much we enforce it at this very moment, but I'd invite you to add a bogus size and give it a shot.
I think it could be interesting to add a `--developer` flag to the core installation command.
I remember we did something for our FW team but they have the whole `package_index_staging.json` generated automatically via an action every time there's a merged PR.
Have you considered that option?

Also I think @silvanocerza can help me remember what was done for the FW team and if we can do something to aid developers whose pre-release workflow is less automated.

Hope he can chime in soon

obra, 2021-10-05:
I have indeed done the auto-generated package on PR/commit thing. It works, but it's a bit clunky. It also means having to publish and store artifacts when we otherwise wouldn't need them.

I did indeed try the bogus size. Sadly, that breaks the classic 1.5 IDE, which does check it. If that had worked with the classic IDE, I'd have suggested documenting a size of 0 or -1 to indicate not checking the size.

Hrm.

ubidefeo, 2021-11-02:
@silvanocerza @obra
I have been giving this a thought, and I highly discourage enabling things at the `index` level.
An alternative (which I believe could be very beneficial to developers and testers) is to add a flag `--unsafe` to some workflows.
This way testers could be instructed to add it (at their own risk) in their configurations as we do with `library.enable_unsafe_install`.
How does that sound?

I believe it's an incredible asset for developers and we don't risk anything since user action is strictly required to achieve this

obra, 2021-11-02:
> @silvanocerza <https://github.com/silvanocerza> @obra <https://github.com/obra>
> I have been giving this a thought, and I highly discourage enabling things at the `index` level.
> An alternative (which [...]
That could work. Would there be a way to enable it within IDE2.0?

ubidefeo, 2021-11-03:
> That could work. Would there be a way to enable it within IDE2.0?
because it's very niche the user could just update the bundled CLI's config file and add the option.
If we take this direction I doubt we'll ever expose it through the UI, but it shouldn't be hard to give instructions to your users.
Does it sound good to you, @obra ?

obra, 2021-11-03:
>

Sure. That sounds great! (And to be clear, for "regular" users, we do absolutely plan to make release builds with SHAs and lengths ;)

rei-vilo, 2023-07-11:
I faced the same issue and found this thread. 

Any news about the options `--unsafe` or similar to `library.enable_unsafe_install`?

Thank you!

cmaglie, 2024-10-29:
Hello :-)

I've made a tentative implementation here: https://github.com/arduino/arduino-cli/pull/2740

obra, 2024-12-24:
Amazing. Thank you so much for this!  This will make it a lot easier to put together 'dev' cores without needing a build step on every commit.

closed 2025-01-14 by cmaglie