arduino/arduino-cli #1473: Build library to .a file, use multiple sketches with .a file

RobMeades, 2021-09-27:
Moved here from [this forum post](https://forum.arduino.cc/t/build-library-to-a-file-use-multiple-sketches-with-a-file/909169).

Not sure if this is my lack of understanding, a [documentation] bug report or an enhancement request.

I would like to ask `arduino-cli` to:

a) build my library, for a given architecture, to a `.a` file,
b) build sketch x, linking it with the above `.a` file.
c) build sketch y, linking it with the above `.a` file.

I can see that the `dot_a_linkage` directive in the library file metadata will tell `arduino-cli` to build the library to an intermediate `.a` file but I can't see how I would tell `arduino-cli` to just build the library for me, so not build a whole sketch but just build the library. The reason I would like to do this as a distinct step is to make sure that the library is built in "its own" way, any conditional compilation is decided by the library builder, not by some random application.

And then, of course, how would I tell it to compile and then link a given sketch using that `.a` file rather than building the library from source (though I guess that might be automatic?)?

I understand that I could do all of this manually (e.g. using an empty `.ino` file for (a) and then copying the resulting library `.a` file somewhere or other), what I'm asking is whether `arduino-cli` can be told to perform these steps by itself; it is a perfectly normal/logical build operation to perform and it looks as though `arduino-cli` should be able to do it, I just can't figure out how.

I suppose that what I really want is a `lib` command which compiles the library for the given platform. Job done.

### Environment

- CLI version `arduino-cli alpha Version: 0.19.0 Commit: 56419ecd Date: 2021-09-02T14:47:35Z`
- OS and platform: Windows 10 Pro 20H2

RobMeades, 2021-09-27:
Turns out I can't even seem to do this manually.

I have two libraries, each of which has the line `dot_a_linkage=true` in their `library.properties` file (one of these is pasted in below).  I have a sketch which includes the headers for the two libraries and I can build that sketch successfully to a binary image:

```
arduino-cli compile --libraries c:\temp\_jenkins_work\libraries\ubxlib --libraries c:\temp\_jenkins_work\libraries\ubxlib_test --fqbn esp32:esp32:esp32doit-devkit-v1 --build-path c:\temp\_jenkins_work\build --build-cache-path c:\temp\_jenkins_work\build c:\projects\ubxlib_priv\port\platform\arduino\app\app.ino
Sketch uses 431953 bytes (32%) of program storage space. Maximum is 1310720 bytes.
Global variables use 24152 bytes (7%) of dynamic memory, leaving 303528 bytes for local variables. Maximum is 327680 bytes.
```

However, the compilation process has only created `.o` files, all of them placed in the `c:\temp\_jenkins_work\build\libraries\src` directory; it has created no `.a` intermediate library file anywhere that I can find.  I've attached the [verbose output](https://github.com/arduino/arduino-cli/files/7239122/no_lib_file_log.txt) from the build (on a second run so as not to have to upload a huge log file with the verbose compilation command-line for each source file in it).

What am I doing wrong?

**library.properties file:**
```
name=ubxlib
version=0.0.0
author=ubxlib@u-blox.com
maintainer=ubxlib@u-blox.com
sentence=Portable C library add-on for MCU and RTOS SDKs to run data and control APIs for embedded applications with u-blox products and services.
paragraph=
category=Communication
url=https://github.com/u-blox/ubxlib
architectures=esp32
dot_a_linkage=true
precompiled=full
includes=ubxlib.h
```

vChavezB, 2021-12-06:
Interesting solution. In my case,[ I am using Cmake](https://github.com/arduino/arduino-cli/issues/1407) to use the arduino SDK to build the library and copy the archive file (.a).

I think this would be a nice feature for developers of Arduino libraries ;)