arduino/arduino-cli #1481: Crash after execute "fatal error: unexpected signal during runtime execution" (macOS 11.6)

judaew, 2021-09-30:
## Bug Report

### Current behavior

`arduino-cli` (tag 0.19.1) built with go 1.17.1. Crashes immediately after executing a binary file:

Crash after execution binary (builded with go 1.17.1):

```console
$ arduino-cli
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0xb01dfacedebac1e pc=0x7fff204ecc9e]
...
```

[arduino-cli-0.19.1-fatal error.txt](https://github.com/arduino/arduino-cli/files/7259691/arduino-cli-0.19.1-fatal.error.txt)

### Environment

- CLI version (~~output of arduino-cli version~~): 0.19.1
- OS and platform: macOS 11.6 x86_64

rsora, 2021-09-30:
@judaew Can you tell us how you installed the CLI?

Are you using `brew` by any chance?

judaew, 2021-09-30:
> Thanks for the detailed bug report!
>
> I think we fixed this bug with #1427 but didn't get through in time for the 0.19.1. May you confitm that the nightly build fix the problem? -> https://arduino.github.io/arduino-cli/0.19/installation/#nightly-builds

@cmaglie This version works fine.

> @judaew Can you tell us how you installed the CLI?
> Are you using brew by any chance?

@rsora I am the maintainer of the [arduino-cli](https://ports.macports.org/port/arduino-cli/details/) port in MacPorts. I wanted to update the port to version 0.19.1 but got this issue.

rsora, 2021-09-30:
Thanks, @judaew for your efforts in porting the CLI on MacPorts! :+1: 

We plan to bump the go version for the build to Go 1.17+ for the next release (meaning `0.20.0`), so you need to be patient and wait for the release to happen in the next weeks :smile_cat: 

Happy coding!

judaew, 2021-09-30:
Thanks so much!

judaew, 2021-10-01:
@rsora Hi, can you review https://github.com/arduino/arduino-cli/pull/1487?

closed 2021-10-01 by per1234