arduino/arduino-cli #1483: updater.enable_notification=false doesn't have the desired effect

RobMeades, 2021-10-01:
## Bug Report
In #1480 I had a problem with the update notification string from `arduino-cli` containing a unicode character which doesn't map to anything by default when run under a Python [3.8] script on Windows 10, causing Python to raise an exception.  The conclusion to that was that I could either sort my goldarned Python/Windows code-page out or disable the notification with:

```
arduino-cli config set updater.enable_notification false
```
While the former was in progress I did the latter but I didn't actually _check_ that it was working because I had already updated my `arduino-cli` version.

My problem is that the notification remains even when I've apparently disabled it, see below.  What am I doing wrong?

### Current behaviour
Dumping my config, you can see that `updater.enable_notification` is `false`.

```
>arduino-cli config dump
board_manager:
  additional_urls: []
daemon:
  port: "50051"
directories:
  data: C:\Users\rob\AppData\Local\Arduino15
  downloads: C:\Users\rob\AppData\Local\Arduino15\staging
  user: C:\Users\rob\Documents\Arduino
library:
  enable_unsafe_install: false
logging:
  file: ""
  format: text
  level: info
metrics:
  addr: :9090
  enabled: true
output:
  no_color: false
sketch:
  always_export_binaries: false
updater:
  enable_notification: false
```

...but when I run, for instance, `arduino-cli version` the "update is available" string still pops out:

```
>arduino-cli version
arduino-cli alpha Version: 0.19.1 Commit: 718bbbf2 Date: 2021-09-29T08:28:11Z

A new release of Arduino CLI is available: 0.19.1 → 0.19.2
https://arduino.github.io/arduino-cli/latest/installation/#latest-packages
```

### Expected behaviour
I would expect to see just:

```
>arduino-cli version
arduino-cli alpha Version: 0.19.1 Commit: 718bbbf2 Date: 2021-09-29T08:28:11Z
```
...no "update is available" string

### Environment
- CLI version: see above
- Windows 10Pro 20H2

per1234, 2021-10-01:
Hi @RobMeades. This is how the feature was intended to work:
https://github.com/arduino/arduino-cli/pull/1416
>This behaviour you can disable it by setting the `updater.disable_notification` config or the env var `ARDUINO_UPDATER_DISABLE_NOTIFICATION` to `true`.
>
>`arduino-cli version` command will always force a check for new versions even if the above config or env var are set.

The configuration key logic [was inverted during the review process](https://github.com/arduino/arduino-cli/pull/1416#discussion_r696733899), so the final behavior is inverted from that description.

This is specific to `arduino-cli version`. The configuration setting will disable the notification for all other commands.

So this isn't a matter of you doing something wrong or a matter of the code not working as intended. We can consider this to be an enhancement request for `updater.enable_notification` to be applied to `arduino-cli version` as well.

RobMeades, 2021-10-01:
Ah! I was just unlucky then :-).

silvanocerza, 2021-10-04:
Yep, @per1234 is right. The version check is always done when running the `version` command. 

In any case as I said in the other issue I strongly suggest you use `--format json` for parsing purposes.

RobMeades, 2021-10-04:
Understood, it's just that we're not parsing this, it is simply for debug output written to a file, we don't actually need/want JSON formatting.

closed 2023-02-01 by cmaglie