arduino/arduino-cli #1486: Initiate process to factor out the CLI Legacy Package

ubidefeo, 2021-10-01:
Arduino CLI embeds some legacy code which should be factored out and/or cleaned up.

- provide an initial assessment of the components to delete
- delete and run tests to check for regressions
- build Arduino Builder at every step and add to Java IDE to test for failure
- repeat until all Legacy is removed

umbynos, 2023-01-30:
This needs to be split in subtasks

umbynos, 2023-05-15:
Related PRs:

- https://github.com/arduino/arduino-cli/pull/2147
- https://github.com/arduino/arduino-cli/pull/2160
- https://github.com/arduino/arduino-cli/pull/2151
- https://github.com/arduino/arduino-cli/pull/2132
- https://github.com/arduino/arduino-cli/pull/2127
- https://github.com/arduino/arduino-cli/pull/2124
- https://github.com/arduino/arduino-cli/pull/2123
- https://github.com/arduino/arduino-cli/pull/2121
- https://github.com/arduino/arduino-cli/pull/2082
- https://github.com/arduino/arduino-cli/pull/2038

umbynos, 2023-09-25:
Completed

closed 2023-09-25 by umbynos