arduino/arduino-cli #1488: Library auto-detection breaks if headers from libraries are needed and g++ does not output English error messages

nicoroeser, 2021-10-02:
## Bug Report

### Current behavior

Tested with `LANG=de_DE.UTF-8`:

Compilation of a project from Arduino IDE or from the command line fails because the necessary include paths (for used libraries) are not appended to the `g++` command. The auto-detection does not work properly because it expects a hard-coded string, which is not present in the compiler output in all locales.

### Expected behavior

Compilation works independently from the current locale.

The code should set `LC_MESSAGES` before executing `g++`.

### Environment

- CLI version (output of `arduino-cli version`):
tested with `arduino-builder-1.5.4`, which links `arduino-cli` as of 2020-06-12T15:26:41-00:00 (3b18571924405b9175cd914d4ef20b07618b92d4).

- OS and platform:
tested on a Gentoo system.

### Additional context

Hard-coded error message is in [`legacy/builder/includes_finder_with_regexp.go`](https://github.com/arduino/arduino-cli/blob/9079f851f48224e5728322e9d27bb3620320eaeb/legacy/builder/includes_finder_with_regexp.go#L38). The [regular expression](https://github.com/arduino/arduino-cli/blob/9079f851f48224e5728322e9d27bb3620320eaeb/legacy/builder/includes_finder_with_regexp.go#L23) may be safe and match in all locales. However, the (more) correct way is to run the compiler in a _known_ locale (`POSIX` or `C` at best, because a machine will be reading the output).

silvanocerza, 2021-10-04:
This is an `arduino-builder` issue right? If that's the case am gonna move it into that repo and probably we'll never fix it.

The `arduino-builder` is deprecated, unmaintained, old and dying. I strongly suggest you use the Arduino CLI instead of the builder.

By the way I couldn't reproduce this issue with `arduino-cli` if you can provide an example to reproduce it I'll be happy to fix it.

nicoroeser, 2021-10-14:
I tried:

```sh
args='-c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/usr/share/arduino/hardware/arduino/avr/variants/standard /tmp/arduino_build_testing/sketch/arduino-cli-issue1488.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE'

for l in de_DE.UTF-8 en_US.UTF-8 ; do
    printf '===== %s =====\n' "$l"
    env "LC_MESSAGES=$l" /usr/bin/avr-g++ $args 2>&1 | cat
    env "LC_MESSAGES=$l" /usr/bin/avr-g++ $args 2>&1 | hexdump -C
    printf \\n
done
```

I added `cat` so that the output would not go to a terminal directly (in order to disable colored output). This should resemble how `g++` is called by `arduino-builder`, I think. I also added hex dumps so that it is clear where space and/or tab characters are (if any).

Output:

```
===== de_DE.UTF-8 =====
/home/user/Arduino/arduino-cli-issue1488/arduino-cli-issue1488.ino:1:10: schwerwiegender Fehler: Servo.h: Datei oder Verzeichnis nicht gefunden
Kompilierung beendet.
00000000  2f 68 6f 6d 65 2f 75 73  65 72 2f 41 72 64 75 69  |/home/user/Ardui|
00000010  6e 6f 2f 61 72 64 75 69  6e 6f 2d 63 6c 69 2d 69  |no/arduino-cli-i|
00000020  73 73 75 65 31 34 38 38  2f 61 72 64 75 69 6e 6f  |ssue1488/arduino|
00000030  2d 63 6c 69 2d 69 73 73  75 65 31 34 38 38 2e 69  |-cli-issue1488.i|
00000040  6e 6f 3a 31 3a 31 30 3a  20 73 63 68 77 65 72 77  |no:1:10: schwerw|
00000050  69 65 67 65 6e 64 65 72  20 46 65 68 6c 65 72 3a  |iegender Fehler:|
00000060  20 53 65 72 76 6f 2e 68  3a 20 44 61 74 65 69 20  | Servo.h: Datei |
00000070  6f 64 65 72 20 56 65 72  7a 65 69 63 68 6e 69 73  |oder Verzeichnis|
00000080  20 6e 69 63 68 74 20 67  65 66 75 6e 64 65 6e 0a  | nicht gefunden.|
00000090  4b 6f 6d 70 69 6c 69 65  72 75 6e 67 20 62 65 65  |Kompilierung bee|
000000a0  6e 64 65 74 2e 0a                                 |ndet..|
000000a6

===== en_US.UTF-8 =====
/home/user/Arduino/arduino-cli-issue1488/arduino-cli-issue1488.ino:1:10: fatal error: Servo.h: No such file or directory
compilation terminated.
00000000  2f 68 6f 6d 65 2f 75 73  65 72 2f 41 72 64 75 69  |/home/user/Ardui|
00000010  6e 6f 2f 61 72 64 75 69  6e 6f 2d 63 6c 69 2d 69  |no/arduino-cli-i|
00000020  73 73 75 65 31 34 38 38  2f 61 72 64 75 69 6e 6f  |ssue1488/arduino|
00000030  2d 63 6c 69 2d 69 73 73  75 65 31 34 38 38 2e 69  |-cli-issue1488.i|
00000040  6e 6f 3a 31 3a 31 30 3a  20 66 61 74 61 6c 20 65  |no:1:10: fatal e|
00000050  72 72 6f 72 3a 20 53 65  72 76 6f 2e 68 3a 20 4e  |rror: Servo.h: N|
00000060  6f 20 73 75 63 68 20 66  69 6c 65 20 6f 72 20 64  |o such file or d|
00000070  69 72 65 63 74 6f 72 79  0a 63 6f 6d 70 69 6c 61  |irectory.compila|
00000080  74 69 6f 6e 20 74 65 72  6d 69 6e 61 74 65 64 2e  |tion terminated.|
00000090  0a                                                |.|
00000091

```

This makes me think that probably function [`findIncludeForOldCompilers`](https://github.com/arduino/arduino-cli/blob/9079f851f48224e5728322e9d27bb3620320eaeb/legacy/builder/includes_finder_with_regexp.go#L33-L44) causes the problem, because it cannot find the line [introducing the error message](https://github.com/arduino/arduino-cli/blob/9079f851f48224e5728322e9d27bb3620320eaeb/legacy/builder/includes_finder_with_regexp.go#L38): it looks for `"fatal error"`.