arduino/arduino-cli #1516: arduino-cli lib upgrade (re-)installs the same libraries multiple times

josefwegner, 2021-10-18:
## Bug Report

### Current behavior

Arduino-cli installs the same libraries over and over:
```
$ arduino-cli lib upgrade
Downloading USBHID@0.63.0...
USBHID@0.63.0 already downloaded
Downloading USBHID@0.63.0...
USBHID@0.63.0 already downloaded
Downloading Servo@1.1.8...
Servo@1.1.8 already downloaded
Downloading Servo@1.1.8...
Servo@1.1.8 already downloaded
Downloading SD@1.2.4...
SD@1.2.4 already downloaded
Installing USBHID@0.63.0...
Already installed USBHID@0.63.0
Installing USBHID@0.63.0...
Already installed USBHID@0.63.0
Installing Servo@1.1.8...
Already installed Servo@1.1.8
Installing Servo@1.1.8...
Already installed Servo@1.1.8
Installing SD@1.2.4...
Already installed SD@1.2.4
```

The cause seems that I have multiple cores installed:
```
arduino-cli core list
ID                  Installed Latest Name
arduino:avr         1.8.3     1.8.3  Arduino AVR Boards
arduino:mbed_nano   2.5.2     2.5.2  Arduino Mbed OS Nano Boards
arduino:mbed_rp2040 2.5.2     2.5.2  Arduino Mbed OS RP2040 Boards
arduino:sam         1.6.12    1.6.12 Arduino SAM Boards (32-bits ARM Cortex-M3)
ATTinyCore:avr      1.5.2     1.5.2  ATTinyCore
eHaJo:avr           2.2.0     2.2.0  eHaJo AVR Boards
MicroCore:avr       2.1.0     2.1.0  MicroCore
MightyCore:avr      2.1.3     2.1.3  MightyCore
MiniCore:avr        2.1.3     2.1.3  MiniCore
sandeepmistry:nRF5  0.7.0     0.7.0  Nordic Semiconductor nRF5 Boards
```
It does not happen if only the arduino:avr core is installed. I removed the `~/.ardunio15` directory:
```
$ arduino-cli update
Updating index: library_index.json.gz downloaded
Updating index: library_index.json.sig downloaded
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
Updating index: package_index.json downloaded
Updating index: package_index.json.sig downloaded
Updating index: library_index.json.gz downloaded
Updating index: library_index.json.sig downloaded
$ arduino-cli upgrade
Downloading missing tool builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
builtin:ctags@5.8-arduino11 installed
Downloading missing tool builtin:serial-discovery@1.3.0-rc1...
builtin:serial-discovery@1.3.0-rc1 downloaded
Installing builtin:serial-discovery@1.3.0-rc1...
builtin:serial-discovery@1.3.0-rc1 installed
Downloading missing tool builtin:mdns-discovery@0.9.2...
builtin:mdns-discovery@0.9.2 downloaded
Installing builtin:mdns-discovery@0.9.2...
builtin:mdns-discovery@0.9.2 installed
$ arduino-cli core list
ID          Installed Latest Name
arduino:avr 1.8.3     1.8.3  Arduino AVR Boards
$ arduino-cli lib upgrade
```
After installing the arduino:mbed_nano core, the command starts installing the same library over and over:
```
$ arduino-cli core install arduino:mbed_nano
Downloading packages...
arduino:openocd@0.11.0-arduino2 downloaded
arduino:arm-none-eabi-gcc@7-2017q4 downloaded
arduino:bossac@1.9.1-arduino2 downloaded
arduino:dfu-util@0.10.0-arduino1 downloaded
arduino:rp2040tools@1.0.2 downloaded
arduino:mbed_nano@2.5.2 downloaded
Installing arduino:openocd@0.11.0-arduino2...
arduino:openocd@0.11.0-arduino2 installed
Installing arduino:arm-none-eabi-gcc@7-2017q4...
arduino:arm-none-eabi-gcc@7-2017q4 installed
Installing arduino:bossac@1.9.1-arduino2...
arduino:bossac@1.9.1-arduino2 installed
Installing arduino:dfu-util@0.10.0-arduino1...
arduino:dfu-util@0.10.0-arduino1 installed
Installing arduino:rp2040tools@1.0.2...
arduino:rp2040tools@1.0.2 installed
Installing platform arduino:mbed_nano@2.5.2...
Configuring platform....
Platform arduino:mbed_nano@2.5.2 installed
$ arduino-cli lib upgrade
Downloading USBHID@0.63.0...
USBHID@0.63.0 downloaded
Installing USBHID@0.63.0...
Installed USBHID@0.63.0
$ arduino-cli lib upgrade
Downloading USBHID@0.63.0...
USBHID@0.63.0 already downloaded
Installing USBHID@0.63.0...
Already installed USBHID@0.63.0
```

### Expected behavior

`arduino-cli lib upgrade` should not install the same libraries over and over when there is no update.

### Environment

- CLI version (output of `arduino-cli version`):
```
$ arduino-cli version
arduino-cli alpha Version: 0.19.3 Commit: 12f1afc2 Date: 2021-10-12T10:15:19Z
```
- OS and platform:
Raspberry Pi OS 32bit (Debian Buster)
```
$ uname -a
Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

### Additional context

None.

ubidefeo, 2021-10-19:
@josefwegner I'll verify this.
most of us have dozens of cores installed but I have not noticed this.
might be something introduced lately after some refactoring.

keep you posted
ubi

closed 2021-12-03 by silvanocerza

josefwegner, 2021-12-07:
Thank you very much. I just verified with arduino-cli 0.20.2-rc1 that this issue has been indeed fixed.

PR #1580 (2021-12-03): Fix `lib upgrade` trying to upgrade core bundled libraries
**Please check if the PR fulfills these requirements**

- [x] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-cli/pulls)
      before creating one)
- [x] The PR follows
      [our contributing guidelines](https://arduino.github.io/arduino-cli/latest/CONTRIBUTING/#pull-requests)
- [x] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] `UPGRADING.md` has been updated with a migration guide (for breaking [... cut]