arduino/arduino-cli #1534: curl error when trying to download to Ubuntu Server 20.04 LTS

travisengh, 2021-10-31:
## Bug Report

### Current behavior

curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/arduino
curl: (23) Failed writing body (481 != 1378)

I'm simply trying to install this to a folder on my ubuntu server and I am receiving the above curl error. I'm new to this, so I am wondering if there is something I am doing wrong? I copied and pasted the downloand link and modified it to suit my needs

### Expected behavior

To download arduino cli to the selected folder on my ubuntu server.

### Environment

- CLI version (output of `arduino-cli version`): Latest
- OS and platform: Ubuntu Server 20.04 LTS

### Additional context

None at the moment

silvanocerza, 2021-11-03:
You're missing `sh` at the end of the line.

This would be the correct command line as per [documentation](https://arduino.github.io/arduino-cli/0.19/installation/#use-the-install-script).
```
curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=~/arduino sh
```

travisengh, 2021-11-03:
I’ll try this today. Thank you!

closed 2021-11-03 by per1234