arduino/arduino-cli #1538: Respect XDG Base Directory Specification

Perlovka, 2021-11-04:
Hi all, 
This is not a bug report, because practice has prove that XDG specs, linux users and common sense are ignored by Arduino project.

So, like in https://github.com/arduino/Arduino/issues/7651, I'll just provide a patch that fixes the issue for linux users seeking to keep their home directories clean.

NOTE: This patch does not make any sense for Windows and Mac users. Sorry.

Some info about XDG base directory specification: 
https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html#introduction

The patch : https://github.com/Perlovka/portage-overlay/blob/1a8ac45f54f40e1bc6e697aeee4f973248277ee2/dev-embedded/arduino-cli/files/arduino-cli-0.19.3-xgd-compliance.patch

Directory layout after applying the patch:
```
~ $ ls -1 ~/.cache/arduino-cli/
staging
tmp
```
```
 ~ $ ls -1 --group-directories-first ~/.local/share/arduino-cli/
libraries
packages
inventory.yaml
library_index.json
library_index.json.sig
package_index.json
package_index.json.sig
```

This patch also fixed configDir path, that is mentioned here:
 https://github.com/arduino/arduino-cli/blob/f7b22f7653ba1094942cee5d67b322763ce00892/configuration/configuration.go#L57

But seems that this value is currently not used elsewhere in the code.

Cheers, may Linux be with you.

silvanocerza, 2021-11-05:
What are the benefits of this change? 
It's a big breaking change, if it doesn't bring any benefit I don't see the value of doing this.

Perlovka, 2021-11-05:
@silvanocerza it's not a change. It's a fix. If you don't understand why, please follow to mentioned arduino issue and read. This has been discussed enough. Feel free to close this issue, it's not for developers, it is for users who understand why.

silvanocerza, 2021-11-05:
It's a change to an existing and established behaviour, a breaking change for all existing users too.
I read the previous discussion and I understand the whys but I'd like to know your opinion and which benefits you think will this bring to users.

If you feel strongly for this feature and think it's important feel free to open a PR, I'll be glad to review it.

Perlovka, 2021-11-05:
@silvanocerza this patch is not change for existing behaviour for linux users, it does exactly opposite, fix the broken for years behaviour of this application.
And this is not a feature, but a must for all applications written with respect to their users.

Linked specification clearly describes default linux/unix application files locations.
And there are reasons for this: distinguish between application configuration, data essential for work, temporary data and so on.
There are also some security reasons for that, but this is not the case now.

For example, all my ~/.config directory is backuped daily, and I don't want to mess with neither finding other essential files to backup nor exclude anything. This directory usually contains only text config files. And I always know where should I go to change any application config - ~/.config/<app_name>

~/.cache directory should contain temporary data that could be deleted at any time without any impact.
~/.local/share should contain application data files, mostly static, like cores, libraries, etc. This data could be also deleted and restored from external sources.

I'm not willing to make a PR now. This issue have been arised many times and always been declined. I made this patch for users who want to keep their user directory clean despite poorly written application.
Also, PR will require significant changes in configuration init, using $XDG_* variables to conform the standard, like I've already done for Arduino app.

Currently this is a dirty hack to drop all the junk from where it is now to appropriate locations.
Updates and fixes could be found in my repo, when they will be available.

brianredbeard, 2022-10-12:
Additionally, I'd take this a step further.

Having just installed the `arduino-cli` utility, it seems absurd that it's choosing an arbitrary path of `~/.arduino15` to be searching considering the current version is 2.x and the legacy version is 1.8.x.

Claiming that this introduces a "breaking change" is picking an odd hill to die on at this point.