arduino/arduino-cli #1540: ESP32 doesn't upload: "Error during Upload: Unknown FQBN: unknown package esp32"

Noxdor, 2021-11-05:
## Bug Report

Instead of uploading the file, it gives the error message shown in the title of this issue.

### Current behavior

The program does compile with no error message, but doesn't upload.

sudo arduino-cli upload -p /dev/ttyUSB0 -b esp32:esp32:featheresp32

There is not much to describe, this is the only error message I am getting, also `--verbose` is not giving me any further explanation.

### Expected behavior

I would expect the cli to upload my program onto the device and to see the device run it.

### Environment

- CLI version (output of `arduino-cli version`):
arduino-cli alpha Version: 0.19.3 Commit: 12f1afc2 Date: 2021-10-12T10:15:19Z
- OS and platform:
Linux 5.14.15-zen1-1-zen #1 ZEN SMP PREEMPT Wed, 27 Oct 2021 23:38:11 +0000 x86_64 GNU/Linux

### Additional context

This is the cores/platforms (I saw there is some discussion which word to use 😉) I have installed (running `arduino-cli core list`):
ID          Installed Latest Name
esp32:esp32 1.0.6     1.0.6  esp32

The board I am having next to me and I want to upload to clearly shows up when listing all boards (running `arduino-cli boad listall | grep feather`):
Adafruit ESP32 Feather          esp32:esp32:featheresp32

Only thing I changed (besides adding the external url for the esp32) in the `arduino-cli.yaml` is the user directory, but that doesn't seem to cause any harm, I already installed two libraries successfully.

per1234, 2021-11-05:
Hi @Noxdor. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there in the dedicated category for Arduino command line tools:

https://forum.arduino.cc/c/software/arduino-cli/89

closed 2021-11-05 by per1234

Noxdor, 2021-11-06:
Hello @per1234, thank you for your reply. I posted this issue here because I do believe it is a bug or at least an unwanted behaviour.
If the FQBN is known during compile, why would it be unknown during upload? Either this is a bug or an error message, that doesn't clearly point to the problem causing this error.
As a user of this cli, it doesn't make sense to my why the FQBN would be known in one step but unknown in another.