arduino/arduino-cli #1541: Q: are executable after `task dist:Linux_64bit` in Release mode?

4ntoine, 2021-11-06:
I've build the `arduino-cli` executable, but it provides low performance at some environment.
I wonder whether executables are compiled in Release or Debug mode. I can see the executable is not stripped, which makes me think it's in Debug mode:
```
file arduino-cli
arduino-cli: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, Go BuildID=Zz60NJg_l4m1AvQOVjSx/dM1hmpgIM7armTx0thNs/_6NAp6OcmjPhXjfQc9rK/r7wrkhfpwfikCFp3Qvwc, not stripped
```

I can definitely strip it, but is there anything to compiled it with optimizations (in Release mode)?
Can/how can i compile it statically?

silvanocerza, 2021-11-08:
Nice idea, I didn't think about it. I guess performance won't change but the binary size seems to be cut quite a bit, at least on Linux where I tested.

My only concern would be if we lose detailed stacktraces in case of crashes. :thinking:

4ntoine, 2021-11-11:
@silvanocerza i'm not sure about the binary tools, but on mobile when the apps are processed the sourcemap (or renamings) is stored separately during the release (even on CI server). Later when crash happens the addresses are processed with the tool that accepts that stored sourcemap and provides a full human-readable stacktrace.

You could smth similar, so the binaries will be faster/smalle, but you will still have the information when smth goes wrong.

umbynos, 2023-02-02:
This is actually a legit request, I think we can take this in consideration. Apparently the stacktrace seems the same between a stripped and a not stripped binary, but the size of the binaries is different