arduino/arduino-cli #1548: ESP32: Wrong library used for "SD.h"

AndreKR, 2021-11-13:
**Describe the bug**
Compiling a sketch that includes `<SD.h>` for ESP32 fails with a "Architecture or board not supported" error because the wrong library is used.

**To Reproduce**
1. This might be a relevant step: I have Arduino 1.8 installed normally and I am running Arduino 2.0 from the ZIP (not installed).
2. Install the ESP32 core via the board manager
3. Set the board to for example "ESP32 Dev Module"
4. Compile this sketch: (The one line is enough to reproduce the issue.)
```c
#include <SD.h>
```
**Actual behavior**
An error happens:
```text
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
Multiple libraries were found for "SD.h"
 Used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\andre\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
Compilation error: Error: 13 INTERNAL: exit status 1
```

**Expected behavior**
Good question. I think the problem is there's an SD library bundled with the IDE and there is one bundled with the core and the one bundled with the IDE is used. Note that I have both Arduino 1.8 and Arduino 2.0 installed and I think the "Used" library comes from Arduino 1.8.

The error happens only in Arduino 2.0, so how does Arduino 1.8 handle this? I would assume the library from the core should take precedence over the library from the IDE?

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: 2.0.0-beta.12

per1234, 2021-11-13:
Hi @AndreKR. Thanks for your report.

This is caused by an intentional change to the Arduino CLI tool that handles the compilation process for the Arduino IDE. The full details on that change, and how it can cause the loss of priority for the platform bundled libraries in rare cases such as the one you experienced, are provided here: https://github.com/arduino/arduino-cli/issues/1292

The fix for that has already been made in the ESP32 boards platform (https://github.com/espressif/arduino-esp32/commit/de66c39f04880e3ae05501d7107762381561fc69), which was released in ESP32 boards 2.0.0. However, you are using an older version of the ESP boards platform (1.0.6), which doesn't have that change.

So you can pick one of the following options for fixing this:

- Use the [Arduino IDE Boards Manager](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-board-manager) to update to ESP32 boards platform version 2.0.0 or later.
- Manually backport the fix by changing line 1 of this file:
  ```
  C:\Users\andre\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD\library.properties
  ```
  from this:
  ```text
  name=SD(esp32)
  ```
  to this:
  ```text
  name=SD
  ```

closed 2021-11-13 by per1234

AndreKR, 2021-11-13:
> So you can pick one of the following options for fixing this:
> * Use the Arduino IDE Boards Manager to update to ESP32 boards platform version 2.0.0 or later.

That was the first thing I tried, but I only got version 1.0.6.

In Arduino IDE 1.8:

[image]

In Arduino IDE 2.0:

[image]

per1234, 2021-11-13:
>That was the first thing I tried, but I only got version 1.0.6.

This is because Arduino IDE 2.x only shows the already installed version of the platform if you don't have the platform's package index URL in your preferences. Arduino IDE 2.x has a completely different preferences storage system from the classic Arduino IDE, so the preferences from the classic Arduino IDE are not used by Arduino IDE 2.x.

Please try this:

1. Select **File > Preferences** from the Arduino IDE 2.x menus.
1. In the "Preferences" dialog that opens, add the following URL to the "Additional Boards Manager URLs" field:
   ```
   https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
   ```
   If there are already URLs in the "Additional Boards Manager URLs" field, separate them with commas (`,`).
1. Close all Arduino IDE 2.x windows.
1. Start the Arduino IDE 2.x
   (restart is required because, due to a bug, Arduino IDE 2.x doesn't otherwise recognize changes to the "Additional Boards Manager URLs" preference)

Now you will be able to update the ESP32 boards platform to the latest version. After doing that, the correct SD library will be used.

AndreKR, 2021-11-13:
That worked.

Using the Preferences dialog also switched the IDE to the light theme. :)

By the way, why is the 2.0 version ignored by Arduino IDE 1.8, I don't see any obvious differences in the JSON?

AndreKR, 2021-11-13:
> By the way, why is the 2.0 version ignored by Arduino IDE 1.8, I don't see any obvious differences in the JSON?

Ah, never mind, I was using a different board URL in Arduino IDE 1.8: `https://dl.espressif.com/dl/package_esp32_index.json`

per1234, 2021-11-13:
>Using the Preferences dialog also switched the IDE to the light theme. :)

This bug is being tracked at https://github.com/arduino/arduino-ide/issues/530

mathemaphysics, 2024-07-05:
Same thing happening for me for esp32 Arduino Nano but in PlatformIO.

So I assume this just isn't going to work.